carla-simulator/carla #5089: make PythonAPI failure

iscanGH, 2022-01-20:


iscanGH, 2022-01-20:
make PythonAPI, fails with the following error, any hint what might have been the reason?

`Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c5-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-0b13b0-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
cp: cannot stat '/home/path/carla/Build/patchelf-install/bin/patchelf': No such file or directory
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1
(base) path@LT6005:~/carla$ whereis patchelf
patchelf: /home/path/anaconda3/bin/patchelf
(base) path@LT6005:~/carla$ make PythonAPI
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c5-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-0b13b0-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
Setup.sh: CARLA version 0.9.13-1-g8854804f-dirty.
Version.h
Setup.sh: Generating CMake configuration files.
LibStdCppToolChain.cmake.gen
LibCppToolChain.cmake.gen
CMakeLists.txt.in.gen
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
-- The C compiler identification is Clang 10.0.0
-- The CXX compiler identification is Clang 10.0.0
-- Could not determine Eclipse version, assuming at least 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Check for working C compiler: /bin/clang-10
-- Check for working C compiler: /bin/clang-10 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/clang++-10
-- Check for working CXX compiler: /bin/clang++-10 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build debug:   OFF
-- Build release: ON
-- Build test:    ON
-- Configuring done
CMake Warning in CMakeLists.txt:
  The build directory is a subdirectory of the source directory.

  This is not supported well by Eclipse.  It is strongly recommended to use a
  build directory which is a sibling of the source directory.

-- Generating done
-- Build files have been written to: /home/path/carla/Build/libcarla-client-build.release
[39/123] Building CXX object LibCarla/.../opendrive/parser/GeometryParser.cpp.o
../../LibCarla/source/carla/opendrive/parser/GeometryParser.cpp:118:21: warning: loop variable 'geo' of type 'const carla::opendrive::parser::Geometry' creates a copy from type 'const carla::opendrive::parser::Geometry' [-Wrange-loop-construct]
    for (auto const geo : geometry) {
                    ^
../../LibCarla/source/carla/opendrive/parser/GeometryParser.cpp:118:10: note: use reference type 'const carla::opendrive::parser::Geometry &' to prevent copying
    for (auto const geo : geometry) {
         ^~~~~~~~~~~~~~~~
                    &
[... 97 lines cut ...]
../../LibCarla/cmake/../source/carla/streaming/low_level/Server.h:25:9: note: candidate found by name lookup is 'carla::streaming::low_level::Server'
  class Server {
        ^
../../LibCarla/source/test/common/test_streaming.cpp:96:3: error: reference to 'Client' is ambiguous
  Client<tcp::Client> c;
  ^
../../LibCarla/cmake/../source/carla/streaming/Client.h:23:9: note: candidate found by name lookup is 'carla::streaming::Client'
  class Client {
        ^
../../LibCarla/cmake/../source/carla/streaming/low_level/Client.h:27:9: note: candidate found by name lookup is 'carla::streaming::low_level::Client'
  class Client {
        ^
4 errors generated.
[117/123] Building CXX object LibCarla...ource/test/client/test_opendrive.cpp.o
../../LibCarla/source/test/client/test_opendrive.cpp:372:42: warning: implicit conversion from 'size_t' (aka 'unsigned long') to 'float' may lose precision [-Wimplicit-int-float-conversion]
      float seconds = 1e-3f * stop_watch.GetElapsedTime();
                            ~ ~~~~~~~~~~~^~~~~~~~~~~~~~~~
../../LibCarla/source/test/client/test_opendrive.cpp:419:42: warning: implicit conversion from 'size_t' (aka 'unsigned long') to 'float' may lose precision [-Wimplicit-int-float-conversion]
      float seconds = 1e-3f * stop_watch.GetElapsedTime();
                            ~ ~~~~~~~~~~~^~~~~~~~~~~~~~~~
2 warnings generated.
[121/123] Building CXX object LibCarla...afficmanager/TrafficManagerLocal.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Util/BuildTools/Linux.mk:124: LibCarla.client.release] Error 1
`

armando-fandango, 2022-02-18:
Getting the same error.

dpresence, 2022-03-31:
same error here

HenryZhuHR, 2022-10-25:
same error here ,how to solve

PatrickFSLin, 2022-11-16:
same error here, no solution?

wutianze, 2022-12-07:
One possible solution is to use clang 8 instead.
Remember to remove "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" from /etc/apt/sources.list, or you will be unable to install clang 8 in 20.04

gufenglcy, 2022-12-12:
got the same error, I am using clang-10, 
According to guide clang-10 is support in 20.04 for carla.

hengcong, 2023-03-02:
I solved this problem by adding 'carla::streaming::low_level' before every ambiguous 'Server' and 'Client'. Actually, it refers to confusion about the definitions of 'Server' and 'Client' since it introduces several definitions in this test_streaming.cpp file.

hengcong, 2023-03-02:
I solved this problem by adding 'carla::streaming::low_level' before every ambiguous 'Server' and 'Client'. Actually, it refers to confusion about the definitions of 'Server' and 'Client' since it introduces several definitions in this test_streaming.cpp file. So, give it a namespace to specify the definition that it needs to call.

Derkai52, 2023-06-21:
same error here, no solution?