carla-simulator/carla #5096: CarlaUE4Editor Crashes after 75% initialization - Windows 10, Carla 0.9.13

Shroculus, 2022-01-20:
Hi everyone,

CARLA version: 0.9.13
Platform/OS: Windows 10
Problem you have experienced: Carla UE4Editor crashes after make launch and 75% of initialization.
What you expected to happen: Expected to successfully open the carlaUE4Editor
Steps to reproduce: Enter command make launch and this results in the fatal error.
Other information (documentation you consulted, workarounds you tried):

**- I Initially had the error**
[BuildCarlaUE4]: Building Unreal Editor... The system cannot find the path specified. make: *** [CarlaUE4Editor] Error 1
 **Resolved with changing the environment variable path from as c:\\UnrealEngine instead of c:\\UnrealEngine\Engine\Binaries\Win64.** 

**- The editor completed 75% of the initialization before the error popped. Screenshot of the error below:**

[image]

**-I tried to open the CarlaUE4 application directly from the Carla root folder but throws the below error:**
LogShaders: Error: The global shader cache file '/content/Engine/GlobalShaderCache-GLSL_150.bin' is missing.
Your application is built to load COOKED content. No COOKED content was found; This usually means you did not cook content for this build.
It also may indicate missing cooked data for a shader platform(e.g., OpenGL under Windows): Make sure your platform's packaging settings include this Targeted RHI.
Alternatively build and run the UNCOOKED version instead_.

**- Now I am even unable to open the Unreal Engine directly from the Unreal engine folder as it throws below error:**
"Plugin ExampleDeviceProfileSelector failed to load because module ExampleDeviceProfileSelector could not be found. Please ensure the plugin is properly installed, otherwise consider disabling the plugin for this project".

Could someone please help?

Thank you in advance!

tiax615, 2022-01-22:
please make sure your content in \Unreal\CarlaUE4\Content\Carla is OK

tiax615, 2022-01-22:
it look like:
[image]

toomuch086, 2022-01-25:
@Shroculus 
Please go to: https://carla.readthedocs.io/en/0.9.13/build_windows/ 
and follow the instructions on >> Part Two: Build CARLA>>Clone the CARLA repository >>Get assets

This command would extract the tar file.
tar -xvzf <assets_file_name>.tar.gz.tar -C C:\path\to\carla\Unreal\CarlaUE4\Content\Carla

I had already extracted the tar file into the correct folder but my CarlaUE4editor crashed at 95%. 
I had the same warning as those ones you displayed above.

Shroculus, 2022-01-27:
> @Shroculus Please go to: https://carla.readthedocs.io/en/0.9.13/build_windows/ and follow the instructions on >> Part Two: Build CARLA>>Clone the CARLA repository >>Get assets.
> 
> The following command would extract the tar file. tar -xvzf <assets_file_name>.tar.gz.tar -C C:\path\to\carla\Unreal\CarlaUE4\Content\Carla
> 
> Although I had already extracted the tar file into the correct folder, the CarlaUE4editor crashed at 95% and I observed the same warning as those ones you attached above. Hopefully, someone can help.

@toomuch086, Thanks a lot! This solved my issue and I could successfully launch the CarlaUE4Editor.

tiax615, 2022-01-27:
> @Shroculus Please go to: https://carla.readthedocs.io/en/0.9.13/build_windows/ and follow the instructions on >> Part Two: Build CARLA>>Clone the CARLA repository >>Get assets.
>
> The following [...]

A lot of time will be spent on shader compilation. Maybe the hardware is not enough, such as memory and cpu.

Shroculus, 2022-01-28:
> A lot of time will be spent on shader compilation. Maybe the hardware is not enough, such as memory and cpu.

Yes, that's true. It took me almost 15-20 mins at 95% initialization to move ahead. Let me know if you need to go over the steps to see what else might be wrong.

toomuch086, 2022-01-30:
> A lot of time will be spent on shader compilation. Maybe the hardware is not enough, such as memory and cpu.

@tiax615 Thank you for the response. Yes, you are correct, the shader compilation requires a lot of time but I wonder why my hardware cannot handle the process. My processor is intel(R) core i7-11370H @ 3.30GHz, RAM is 8GB. GPU is NVIDIA GeForce RTX 3060.

toomuch086, 2022-01-30:
> Yes, that's true. It took me almost 15-20 mins at 95% initialization to move ahead. Let me know if you need to go over the steps to see what else might be wrong.

@Shroculus. After several attempts, I managed to launch Carla but unfortunately, it crashed and it would not launch again. I intend to increase my RAM size and try again. Thank you for the support.

closed 2022-02-03 by Shroculus

longshujing, 2022-05-17:
@toomuch086 
hi，toomuch86
I have the same problem as you. Have you solved it? Is it the hardware?

FBanani, 2022-07-03:
hi
I have some issue after making launch
1: when carlaUE4 initialize 75% it shows a message with some errors about failed to find.
[image]

then I press ok. the carlaUE4_unreal editor open but it shows message logs with many failed to load.
[image]

I close it and press play but it shows " One or more blueprints has an unresolved compiler error, are you sure you want to Play in Editor?"
[image]

I press play in editor and after a time the CarlaUE4 is closed with the box "the UE4_carlaUE4 editor has crashed" fatal error .
[image]

what should I do?

xuq9908, 2023-05-09:
> One or more blueprints has an unresolved compiler error, are you sure you want to Play in Editor?

Hi, I have the same problem as you. How did you solve that?

xuq9908, 2023-05-09:
> hi I have some issue after making launch 1: when carlaUE4 initialize 75% it shows a message with some errors about failed to find. [image]
>
> then I press ok. the carlaUE4_unreal editor open but it [...]

Hi, I have the same problem as you. How did you solve that?

FBanani, 2023-05-09:
> unresolved compiler error, are you sure you want to Play in Editor?" [image:
> Capture2]

Actually I do not remember what exactly but I remember I search a lot.
Sorry