carla-simulator/carla #5097: Clang-8 build problems (broken packages)

chrninja, 2022-01-20:
https://github.com/carla-simulator/carla/issues/2076#issue-489936022

Facing the same issue as described here except with clang-8. I tried the route with aptitude and purging clang-8 and other packages but I am not sure how to solve this problem.

CARLA version: 09.13
Platform/OS: Ubuntu 20.04

Following the Linux source build to first install all required packages and the same compiler for Carla and UE4

https://carla.readthedocs.io/en/0.9.13/build_linux/

chrninja, 2022-01-21:
I resolved the problem with `sudo add-apt-repository -r "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main"` since I don't need the xenial deb files on 20.04 (focal fossa) according to  [this](https://github.com/carla-simulator/carla/pull/2787) I have clang-8 now on my system but when I compile UE4 it uses v17-clang10.01.

I think I will run into problems when building from Carla later since both need to use clang-8.

chrninja, 2022-01-22:
No problems with clang thereafter. What I needed to fix was change boost version 1.72 to version 1.69.

closed 2022-01-22 by chrninja

hdp-alt, 2022-08-04:
Try:
sudo add-apt-repository -r "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" &&
sudo apt install clang-8