carla-simulator/carla #5099: Map ingestion from Road Runner to Unreal engine

werewolfdev, 2022-01-21:
HI Team,

I have exported a simple map from Road runner and kept the .fbx and .xodr files in the import folder of CARLA build from source. I tried make import from the CARLA root directory and I am getting the following error. My default python is 2.7 , so I am running all the individual scripts using python3.6 , my egg is also same (3.6). Could anyone kindly let me know any ideas on how to solve this issue. I have tried importing the .fbx directly by using the plugins of road runner , but then the Unreal engine crashes all the time or doesn't respond

CARLA version:0.9.13 (latest)
Unreal Engine : 4.26
Platform/OS:Ubuntu 18.04
What you expected to happen: Map ingestion into UE4

Error:
BuildPythonAPI.sh: Success!
Copying "/home/schemba/carla_new/carla/Import/town_test.xodr" to "/home/schemba/carla_new/carla/Util/DockerUtils/dist/town_test.xodr"
Commandlet: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor /home/schemba/carla_new/carla/Unreal/CarlaUE4/CarlaUE4.uproject -run=ImportAssets -importSettings="/home/schemba/carla_new/carla/importsetting.json" -nosourcecontrol -replaceexisting
Copying "/home/schemba/carla_new/carla/Import/town_test.fbx" to "/home/schemba/carla_new/carla/Util/DockerUtils/dist/town_test.fbx"
/bin/sh: 1: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor: not found
Copying "/home/schemba/carla_new/carla/Import/town_test.xodr" to "/home/schemba/carla_new/carla/Unreal/CarlaUE4/Content/test_map/Maps/town_test/OpenDrive/town_test.xodr"
Commandlet: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor /home/schemba/carla_new/carla/Unreal/CarlaUE4/CarlaUE4.uproject -run=MoveAssets -PackageName=test_map -Maps=town_test 
/bin/sh: 1: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor: not found
Commandlet: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor /home/schemba/carla_new/carla/Unreal/CarlaUE4/CarlaUE4.uproject -run=PrepareAssetsForCooking -PackageName=test_map -OnlyPrepareMaps=1
/bin/sh: 1: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor: not found
Commandlet: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor /home/schemba/carla_new/carla/Unreal/CarlaUE4/CarlaUE4.uproject -run=LoadAssetMaterials -PackageName=test_map
/bin/sh: 1: :/home/schemba/CARLA_0.9.12/UnrealEngine_4.26/Engine/Binaries/Linux/UE4Editor: not found
WARNING: cannot parse georeference: ''. Using default values. 
Traceback (most recent call last):
  File "get_xodr_crosswalks.py", line 21, in <module>
    import carla
  File "/home/schemba/carla_new/carla/PythonAPI/carla/dist/carla-0.9.13-py3.6-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
    
  File "/home/schemba/carla_new/carla/PythonAPI/carla/dist/carla-0.9.13-py3.6-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/home/schemba/carla_new/carla/PythonAPI/carla/dist/carla-0.9.13-py3.6-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
ImportError: /home/schemba/.cache/Python-Eggs/carla-0.9.13-py3.6-linux-x86_64.egg-tmp/carla/libcarla.cpython-36m-x86_64-linux-gnu.so: undefined symbol: PyUnicode_AsUTF8String

xiaoqingnianbudadun, 2022-01-21:
Did you had build UE4 use command lilk:
cd $UE4_ROOT && \
  ./Setup.sh && \
  ./GenerateProjectFiles.sh && \
  make
there is no UE4Editor in your project

werewolfdev, 2022-01-21:
Hi @xiaoqingnianbudadun I ahve followed all the instructions from here: https://carla.readthedocs.io/en/latest/build_linux/ . Also, i have run this command . I have run scenarios and so on with UE4 editor already, just the map ingestion, import operation fails. I think this is linked to python version probably since I have default py2.7 , I am using the egg 3.6 , and using python3.6 command to run all the scripts

werewolfdev, 2022-01-22:
@fabianoboril Hi sir, do you have any insights on this issue

werewolfdev, 2022-01-22:
The UE4 Root is not properly read , I did export for the session and everything seems normal but UE doesnt open up as of now

werewolfdev, 2022-01-26:
Hi @bernatx Could you kindly provide any insights on the same. i have exported the files ,put into make import folder and did make import with package name. Python API build was successful and no errors could be seen in the terminal , carla unreal engine editor just didnt load and these log messages showed up after a couple of minutes

bernatx, 2022-01-26:
Hi,
The **make import** command just cook the map into the unreal engine, and generate one (or several) .umap into the carla content. At this time, you can start unreal (with **make launch-only** for example) and load your recently imported map.

So, what make import does is just cooking your geometry from .fbx (inside the /import/ folder) to .umap (into /carla/Unreal/CarlaUE4/Content/[package] folder).

In the process of importing, it is normal that several mini unreal engines start and finish because they are executing different commandlets, one per run.

werewolfdev, 2022-01-26:
thanks for the quick reply. so after performing make import I should run **make launch-only**  for loading the editor with that map if Iam not wrong

werewolfdev, 2022-02-03:
So the way to do it is to export .fbx and .xodr files from Road runner to CARLA Import folder , run make import and wait for it finish running and then run make launch-only and then once the carla UE4 Editor opens up open File->Open Level-> and select the generated package. I hope this could be properly mentioned in the documentation. I found it out from a video from youtube published by CARLA itself but would love to have it in documentation as well so as to save time for other users

closed 2022-02-03 by werewolfdev