carla-simulator/carla #5100: Getting the world 3d coordinates from mouse click in pygame window in CARLA

kanji95, 2022-01-21:
I am having trouble getting the 3D world coordinates in CARLA. Basically, I use mouse click to select the destination point (pixel coordinate) in the front-view RGB camera image and get the depth of that point through a depth sensor. Now, I get the 3D coordinate using the following,
`pos_3d = np.linanlg.inv(K) @ pos_2d * depth.  # pos_2d = [u, v, 1]
  pos_3d = T_cam @ pos_3d`

Here, K is the intrinsic camera matrix and T_cam is Camera extrinsic transformation matrix. However, the final point I get after the above transformation is not correct, it is basically a point behind the vehicle's starting position. I am unable to figure out what I am doing wrong.

brscholz, 2022-02-07:
Correct me if I'm wrong, but I think that won't work anyway because you have zero clue as to where along its way the ray that is represented by the pixel met a surface? How do you want to extract that depth information from the image?

DanyThodms, 2023-05-31:
We encountered the same problem and ultimately found out that it was because we used a remote connection that caused a mouse mapping problem, which would not have occurred on the computer where the project was located.

kanji95, 2023-06-12:
I was able to resolve this issue by applying the additional transformation of CARLA to world coordinates.

closed 2023-06-12 by kanji95