carla-simulator/carla #5107: Custom Controller: Pedestrian unable to climb to sidewalks from road

adhocmaster, 2022-01-25:
CARLA version: 0.9.13
I made a custom map in Road Runner. WalkerAIController can crossroads, walk on sidewalks, everything is smooth. Now I am writing my own controller in python. Everything is good, except for one problem. When the pedestrian is on-road it cannot climb up a sidewalk. For the time being, I am jumping onto the sidewalk when the pedestrian is close to a sidewalk. Ideally, I would like it to climb up smoothly. Can anyone guide me to do it, please?

LucasFKobernic, 2022-01-31:
I had the same problem in the past. What solved my problem was:

1. In the Unreal Engine editor select the curb;
2. Under Collision-> Collision Presets -> Choose "NoCollision"
3. save it

After that my pedestrian was able to climb up the sidewalk.
[image]

closed 2023-06-02 by stale[bot]

adhocmaster, 2023-07-31:
Need another solution which does not required rebuilding existing Carla maps

zhz03, 2024-05-11:
HI, @adhocmaster  Did you solve this issue?

adhocmaster, 2024-05-20:
I wrote my custom algorithm to avoid the collision as rebuilding the maps were not an option. Basically, I calculated whether there would be a collision soon. If true, I made the pedestrian jump onto the sidewalk.