carla-simulator/carla #5111: RuntimeError: trying to create roc server for traffic manager; but the system failed to create because of bind error

Zhenlin-Xu, 2022-01-28:
CARLA version: 0.9.13 RSS
Platform/OS: Ubuntu18.04 LTS
Problem you have experienced: **RuntimeError: trying to create roc server for traffic manager; but the system failed to create because of bind error**
What you expected to happen: I want to write a self-defined gym class of Carla for deep reinforcement learning. When I plan to initialise the traffic managers in the __init__(self, ...) function, I come up with this error. I also tried to run the automatic_control.py in the path `CARLA_DIRECTORY/PythonAPI/example`, it failed too with the same error. However, when I downloaded the package and ran the automatic_control.py file at the first time, the program ran successfully. 
Other information (documentation you consulted, workarounds you tried): I referred to other issues with the same problem, i.e. check the port and process. However, I am not sure about the detailed implementation of this in my computer.

jackbart94, 2022-02-01:
It could be for a series of problems in your code. Can you give us some way to replicate this issue?

Zhenlin-Xu, 2022-02-06:
I reinstalled the binary and it worked fine. Thx for your reply!

closed 2022-02-06 by Zhenlin-Xu

zkailinzhang, 2022-08-01:
same error

fah870599552, 2023-02-21:
> I reinstalled the binary and it worked fine. Thx for your reply!

hello, I get the same problem, How did you solve it ? with much thanks

fah870599552, 2023-02-21:
> same error

me too, Did you solve it?

Zhenlin-Xu, 2023-02-21:
> > I reinstalled the binary and it worked fine. Thx for your reply!
> 
> hello, I get the same problem, How did you solve it ? with much thanks

I remember I just downloaded a new binary package and it worked well. It is kinda of weird, to be honest.

fah870599552, 2023-02-21:
> > > I reinstalled the binary and it worked fine. Thx for your reply!
> > 
> > 
> > hello, I get the same problem, How did you solve it ? with much thanks
> 
> I remember I just downloaded a new binary package and it worked well. It is kinda of weird, to be honest.

EXCEUSE ME COULD YOU TELL ME WHICH BINARY PACKAGE IT IS. THANKS

Zhenlin-Xu, 2023-02-21:
> EXCEUSE ME COULD YOU TELL ME WHICH BINARY PACKAGE IT IS. THANKS

[image]

just download the package that fits your OS. 😀