carla-simulator/carla #5135: ImportError: cannot import name 'image_converter' from 'carla' - Carla 0.9.13, Windows 10, Human_Control code - G920 steering wheel

Shroculus, 2022-02-03:
Hi everyone!

**CARLA version:** 0.9.13
**Platform/OS**: Windows 10
**Problem you have experienced:** When I try to execute the human_control.py it throws me the error: "ImportError: cannot import name 'image_converter' from 'carla'". 
**What you expected to happen:** To successfully setup G920 steering wheel with manual controls.
**Steps to reproduce:** 

1. Imported and saved the file human_control.py in the path PythonAPI/examples from https://github.com/rohitgajawada/carla/blob/intern/PythonClient/human_control.py
2. Ran the human_control.py from the command prompt at example folder.
3. Simultaneously launched the CarlaUE4Editor and was in the play mode. 

**Other information (documentation you consulted, workarounds you tried):**

1. Tried to install the .egg in the PythonAPI/carla/dist folder using easy_install, pip and pip3 it results in error "'easy_install' is not recognized as an internal or external command operable program or batch file." 
2. With pip & pip3 it throws error "ERROR: Invalid requirement: 'C:\\Users\\hcelab\\carla\\PythonAPI\\carla\\dist\\carla-0.9.13-py3.8-win-amd64.egg' Hint: It looks like a path. The path does exist."
3. My setuptools are all updated and the requirement is satisfied too. 
4. I can even see easy_install.py under the setuptools/commands folder.

[image]

**The full error w.r.t Carla is shown below:**
[image]

Could someone please help me or at least point me in the right direction. Any help is much appreciated.

Thanks in advance.

closed 2023-06-02 by stale[bot]

i-kj, 2023-10-19:
Hello, 

I am also facing the same issue. Did you manage to fix it?

ggambi0505, 2024-02-08:
where can i find carla\PythonAPI\carla\dist\carla-0.9.13-py3.8-win-amd64.egg ????