carla-simulator/carla #5145: Install Ros bridge in Virtual Machine with [bridge-1] ModuleNotFoundError: No module named 'carla error

kennyS99, 2022-02-06:
Thanks for contributing to CARLA!

CARLA version: host: 0.9.13 VM:0.9.13
Python version: host:3.7 VM:3.8 
Platform/OS:windows10 and ubuntu 20.0.4lLTS virtual machine
Ros2 version: foxy

I am running Carla and Unreal Engine on windows 10 host machine and running the ros-bridge on a Ubuntu virtual machine(VM ware).

I have already tested the Carla and unreal engine on my host machine, they worked very well.

I have already tried the solution that I can find online.
Install windows version Carla on host machine and install ros2, Carla and Ros bridge.
use command with the complete path to the .egg file which is installed on ubuntu VM.
export PYTHONPATH=$PYTHONPATH:path/to/carla/PythonAPI/<your_egg_file>
and change the localhost of carla_ros_bridge.launch.py to the host IP address.
The host machine IP address has been tested successfully.

checked the CARLA library can be imported correctly by the following command and received the Success message.
python3 -c 'import carla;print("Success")' # python3

Try to run ros bridge by the official tutorial:
    # Option 1, start the basic ROS bridge package
    ros2 launch carla_ros_bridge carla_ros_bridge.launch.py
and received the following error message:

kenny@kenny-virtual-machine:~$ ros2 launch carla_ros_bridge carla_ros_bridge.launch.py
[INFO] [launch]: All log files can be found below /home/kenny/.ros/log/2022-02-02-23-33-31-983942-kenny-virtual-machine-7645
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [bridge-1]: process started with pid [7647]
[bridge-1] Traceback (most recent call last):
[bridge-1] File "/home/kenny/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge", line 11, in
[bridge-1] load_entry_point('carla-ros-bridge==0.0.0', 'console_scripts', 'bridge')()
[bridge-1] File "/home/kenny/.local/lib/python3.8/site-packages/pkg_resources/init.py", line 489, in load_entry_point
[bridge-1] return get_distribution(dist).load_entry_point(group, name)
[bridge-1] File "/home/kenny/.local/lib/python3.8/site-packages/pkg_resources/init.py", line 2861, in load_entry_point
[bridge-1] return ep.load()
[bridge-1] File "/home/kenny/.local/lib/python3.8/site-packages/pkg_resources/init.py", line 2461, in load
[bridge-1] return self.resolve()
[bridge-1] File "/home/kenny/.local/lib/python3.8/site-packages/pkg_resources/init.py", line 2467, in resolve
[bridge-1] module = import(self.module_name, fromlist=['name'], level=0)
[bridge-1] File "/home/kenny/carla-ros-bridge/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/bridge.py", line 24, in

[bridge-1] import carla
**[bridge-1] ModuleNotFoundError: No module named 'carla'**
[ERROR] [bridge-1]: process has died [pid 7647, exit code 1, cmd '/home/kenny/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r _node:=carla_ros_bridge --params-file /tmp/launch_params_v997htaz --params-file /tmp/launch_params_vfxwwxsz --params-file /tmp/launch_params_cvtstc9q --params-file /tmp/launch_params_bc7eeaob --params-file /tmp/launch_params_uw098dmw --params-file /tmp/launch_params_523rl7y2 --params-file /tmp/launch_params_cwipl593 --params-file /tmp/launch_params_dhrmfak5 --params-file /tmp/launch_params_tz4vm998 --params-file /tmp/launch_params_pvtp_j7 --params-file /tmp/launch_params_pa21hboq'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system

Anyone can provide help?
Thank you so much.

MunchZkin, 2022-02-16:
I am working on a similar setup 
try `python3.8 -m pip install carla` or use python3.7 on your VM using this [Installing Python 3.7 on Ubuntu with Apt](https://linuxize.com/post/how-to-install-python-3-7-on-ubuntu-18-04/)

Also I am not sure if the carla-ros-bridge is updated to work well with 0.9.13, try a version lower probably if you face any issues

kennyS99, 2022-02-22:
> I am working on a similar setup try `python3.8 -m pip install carla` or use python3.7 on your VM using this [Installing Python 3.7 on Ubuntu with [...]

I have successfully solved the problem by changing the PATHONPYTHON in my bash file.
export CARLA_ROOT=/home/kenny0407/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.12-py.12-py3.8-linux-x86_64.egg

I have another problem when I want to run the official Carla ros2 bridge tutorial code like below:
    # Option 1, start the basic ROS bridge package
    ros2 launch carla_ros_bridge carla_ros_bridge.launch.py
which change my window from the default map to town 1, and nothing happened.

    # Option 2, start the ROS bridge with an example ego vehicle
    ros2 launch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch.py

when I try to run the option2, I have the following error
[bridge-1] [INFO] [1645506581.235486667] [carla_ros_bridge]: Created Actor(id=194)
[bridge-1] [INFO] [1645506581.235834651] [carla_ros_bridge]: Created Actor(id=195)
[bridge-1] [INFO] [1645506581.236159862] [carla_ros_bridge]: Created Actor(id=196)
[ERROR] [bridge-1]: process[bridge-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [bridge-1]: sending signal 'SIGTERM' to process[bridge-1]
[ERROR] [bridge-1]: process has died [pid 55557, exit code -15, cmd '/home/kenny0407/carla-ros-bridge/install/carla_ros_bridge/lib/carla_ros_bridge/bridge --ros-args -r __node:=carla_ros_bridge --params-file /tmp/launch_params_bxdop73n --params-file /tmp/launch_params_t6bfylmn --params-file /tmp/launch_params_m2tf73lc --params-file /tmp/launch_params_ihhzi7ly --params-file /tmp/launch_params_krlsl3n_ --params-file /tmp/launch_params_cks_8d0k --params-file /tmp/launch_params_55ox3cqd --params-file /tmp/launch_params_i6i692tu --params-file /tmp/launch_params_9oao6yn8 --params-file /tmp/launch_params_shz8uxtx --params-file /tmp/launch_params_9pncoa0z'].
[INFO] [launch]: process[bridge-1] was required: shutting down launched system

do you have any suggestions to solve this one? 
Thank you very much.

vttitta, 2022-07-25:
Were you able to solve this and run the simulator in a different environment than the code/bridge? Having the exact same problem here

closed 2024-05-17 by MarcelPiNacy-CVC