carla-simulator/carla #5147: Filtering the vehicles and walkers together into a list

werewolfdev, 2022-02-07:
Hi team , i would like to generate a player list with all the vehicles and the walkers together. The filter functionality associated with get_actors enable to filter only one type if I understood correctly. As the ActorList object doesnot have append functionality I cannot append the earlier generated list with one more filter('walker.*') as well. Could you kindly let em know if there is any way of filtering these objects together to an object.
CARLA version: 0.9.13
Platform/OS:Ubuntu 18.04

closed 2022-02-07 by werewolfdev

MMNavetty, 2022-07-04:
hello, could you provide the solution please ? i'm having the same problem here.
Thanks in advance,
MM

yhh-IV, 2024-03-12:
the same issue....