carla-simulator/carla #5154: Carla 0.9.13 Ran out of memory allocating 18446744068997660672 bytes with alignment 65536 Signal 11 caught.

ucbmpc2021, 2022-02-10:
Specs:
CARLA package 0.9.13
Unreal Engine 4.24 and 4.26 installed on my computer
Ubuntu 18.04.6

When launch CarlaUE4.sh after downloading release package, I get the following error:
`4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
LowLevelFatalError [File:Unknown] [Line: 1031] 
Ran out of memory allocating 18446744068997660672 bytes with alignment 65536
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=131122 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=262312 
Malloc Size=131160 LargeMemoryPoolOffset=393496 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
`

alexx-km, 2022-02-17:
I've got the same issue, but I don't have both Unreal Engines installed. Have you found a solution in the meantime?

ucbmpc2021, 2022-02-24:
So because of some other reasons I restored my whole Linux system to an initial stage and this problem is gone. I think one possible cause for this was: I cooked a map using Carla docker 0.9.9, imported it into this 0.9.13, and launched this map, and this somehow messed up my Unreal Engine memory.

alexx-km, 2022-02-25:
@ucbmpc2021 Alright, thank you! I also restored a previous stage (by reinstalling CARLA) and the issue is gone, too!

abdooo786, 2022-04-22:
i don't know why , but it worked with me using "./CarlaUE4.sh -opengl13" 
ubuntu 18.04
0.9.11
ryzen 5 3400G

closed 2024-05-17 by MarcelPiNacy-CVC