carla-simulator/carla #5155: customized map using pre existing map

pushkar4501, 2022-02-10:
CARLA version:  0.9.10
Platform/OS:  linux
unreal engine : 4.24
 
Hello everyone, I am trying to customize the Town map already present in carla, using other carla assets. I have successfully custmized the map using real engine editor directly as I just can use most of the assets from carla town map. The customized map file name I can see is .uasset. 
Based on other recommendation, i created the .FBX and .xodr format and saved it the openDrive Folder as well, still i am unable to acess the latest customized map using python api. 
Please guide me to how to create and open the custmized map through python api.

koach87, 2022-02-14:
Sorry, may i ask you how you generating the fbx and OpenDrive file ?

pushkar4501, 2022-02-18:
can follow this issue:
https://github.com/carla-simulator/carla/issues/4514

koach87, 2022-02-18:
Thanks so much!
I'll take a look.

closed 2023-06-02 by stale[bot]