carla-simulator/carla #5156: Unreal 4.27 support

den1den, 2022-02-10:
Hi,

First of all, I love the project. We're using it to create a driving simulator (with a moving base and 3 attached beamers). In order to properly wrap the projection of unreal to the beamers we are making use of [nDisplay](https://docs.unrealengine.com/4.27/en-US/WorkingWithMedia/IntegratingMedia/nDisplay/Overview/). We achieved the desired result with 4.26 (eventually) but Unreal 4.27 supports this technology with significantly better tool support. 

Are there plans on going to 4.27 in the coming year?

We're using:
CARLA 0.9.13 on Windows 10 and Ubuntu 20

eliabruni, 2023-01-28:
Teaser: At [Saivvy](https://github.com/saivvy) we are about to release 4.27 support for CARLA - stay tuned!

petebaker17, 2023-02-21:
This sounds great @eliabruni! Keep us in the loop on progress if you can!