carla-simulator/carla #5163: 8 fps when running in server mode

ValeriaProiettiDante, 2022-02-14:
Hello everyone!

I have an Ubuntu system with an NVIDIA 3080ti with CARLA and Unreal Engine 4.26.

When CARLA is not in server mode, I have 60fps, but when I run the server, the fps go down to 8.

I have already tried to uncheck the "Use less CPU" parameter in the Unreal editor.

Anyone have any suggestions?

Thank you

MunchZkin, 2022-02-16:
I personally don't use Carla on Linux due to poor GPU support. But when you say "run the server" what do you mean?
Generally, when you have more actors in Carla you usually tend to loose fps, and these actors can be cameras, other vehicles/ walkers, streetlamps etc. So try restricting these actors needed specifically for you use case.

ValeriaProiettiDante, 2022-02-17:
Thanks @MunchZkin for reply.

For "run the server", I mean press play on UnrealEngine editor and it starts on server mode for CARLA simulator and, in my case, ROS2.

closed 2023-06-02 by stale[bot]