carla-simulator/carla #5167: RuntimeError: Spawn failed because of collision at spawn position

marco-didom, 2022-02-15:
i can't run create_sumo_vtypes.py which is located in carla \ co-simulation \ sumo \ util folder.

I get this error message: RuntimeError: Spawn failed because of collision at spawn position

Consequently I am unable to generate any vtypes

werewolfdev, 2022-02-16:
Hi @marco-didom : RuntimeError: Spawn failed because of collision at spawn position Used to get these errors when you have multiple actors trying to spawn at the same position

marco-didom, 2022-02-16:
hi @werewolfdev , thanks for your reply. 

how do i avoid it? create_sumo_vtypes.py is provided as an example by CARLA, I didn't create it. I just can't run the simulation shown in the examples on the carla simulator site

werewolfdev, 2022-02-16:
I dont have experience with scripts regarding sumo, but when I used to create scenarios using scenario runner with multiple targets, I just made sure the actors are not so close to each other while providing their spawn positions .  Try looking at how the vehicles are spawned in these scripts if its multi target based scripts.

gabrielzemmit, 2023-01-25:
@marco-didom i just ran into the exact same problem. have you found any solution to this?