carla-simulator/carla #5174: Can use City: Skylines game build maps for CARLA?

TianhangWang, 2022-02-17:
Sorry to bother you, I am wondering whether the GAME of `Cities: Skylines` which can build your own world can co-work with CARLA. I mean I have read the official docs about how to build custom map by using `.fbx` and `.xdor` which can be exported by different softwares. So does CARLA have the interface with such game `Cities: Skylines` which can make custom map more easier?

There is the link about GAME of `Cities: Skylines`: https://store.steampowered.com/app/255710/Cities_Skylines/

TianhangWang, 2022-02-19:
I see, thank you for your replying! Meanwhile, I want to use CARLA to build the scene of harbour to simluate automotive test. So, does CARLA can realize interactive simulation? I mean when cargo trunk goes to  the the specific location, the big crane in the harbour will pack trunk's cargo automatically. Could you tell me how to realize that?

RocaPiedra, 2022-02-28:
> I see, thank you for your replying! Meanwhile, I want to use CARLA to build the scene of harbour to simluate automotive test. So, does CARLA can realize interactive simulation? I mean when cargo [...]

I'm not an expert on Carla but I think that won't be possible. Carla is oriented towards autonomous driving so it won't give you the tools to do those tasks. Maybe with Unreal Engine you can prepare an environment/map that can do that for you and then launch Carla to perform the driving-related tasks.

TianhangWang, 2022-02-28:
Thank for your reply.