carla-simulator/carla #5179: autopilot vehicle drives off road

makaveli10, 2022-02-18:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04.3 LTS
Autopilot vehicle drives off road at the next junction or turn; no matter where it is spawned. I used 0.9.11 before this and everything works just fine.
Steps to reproduce: 
- Run carla server
- Create ego vehicle from client.
```
import os
import carla
import random

# connect client to carla server
import time
start = time.time()
while True:
    try:
        print("Waiting for CARLA")
        client = carla.Client('localhost, 2000)
        client.set_timeout(10.0)
        world = client.get_world()
        break
    except:
        import traceback
        traceback.print_exc()
        time.sleep(5)
        pass

# spawn an ego-vehicle randomly
spawn_points = world.get_map().get_spawn_points()
blueprints_vehicles = world.get_blueprint_library().filter("vehicle")
ego_transform = spawn_points[random.randint(0, len(spawn_points) - 1)]

blueprints_vehicles = [x for x in blueprints_vehicles if int(x.get_attribute('number_of_wheels')) == 4]

# let CarlaViz know this vehicle is the ego vehicle.
blueprints_vehicles[0].set_attribute('role_name', 'ego') # or set to 'hero'
batch = [carla.command.SpawnActor(blueprints_vehicles[0], ego_transform).then(carla.command.SetAutopilot(carla.command.FutureActor, True))]
results = client.apply_batch_sync(batch, True)
if not results[0].error:
    ego_vehicle = world.get_actor(results[0].actor_id)

while True:
    world.tick()
```

This code works as expected in 0.9.11. For 0.9.12+ the autopilot drives off road and is eventually destroyed.

jackbart94, 2022-02-21:
Are you using synchronous mode when running the simulation?

makaveli10, 2022-02-23:
No @jackbart94 . The code above is all that I used.

jackbart94, 2022-02-23:
Asynchronous mode has been sort-of deprecated from TM in 0.9.12 and onwards. We always expect the simulation to be in synchronous mode.

makaveli10, 2022-02-24:
Hi @jackbart94 I tired the synchronous mode i.e.
```
settings = world.get_settings()
settings.synchronous_mode = True # Enables synchronous mode
world.apply_settings(settings)
```
In 0.9.11 the off road is resolved when using this, in 0.9.12 the autopilot is driving very naively it seems on edge of lanes accelerating and decelerating very frequently. As for 0.9.13, it drives road on the next junction.

jackbart94, 2022-02-24:
Could you also set fixed_time_step to 0.05, as per the tutorial, while setting synchronous mode?

makaveli10, 2022-03-02:
@jackbart94 Even after setting ```fixed_time_step=0.05``` autopilot cannot stay on track in version ```0.9.13```. But the issue is fixed in previous version i.e ```0.9.12```.

alexkutsan, 2022-10-12:
I have a very similar issue https://github.com/carla-simulator/carla/issues/5707#issuecomment-1271871350

ronyshaji, 2022-10-28:
> I have a very similar issue [#5707 (comment)](https://github.com/carla-simulator/carla/issues/5707#issuecomment-1271871350)

Did you able to fix the issue ?

makaveli10, 2023-03-14:
@ronyshaji yes... by using different traffic manager port for different clients.

closed 2023-03-14 by makaveli10

hexrefugee, 2023-05-29:
> @ronyshaji yes... by using different traffic manager port for different clients.
Is that means :
terminal 1: ./CarlaUE4.sh -carla-rpc-port=4000
terminal 2: python3 spawn_npc.py --port 4000 --tm-port 4050 # TM-Server
terminal 3: python3 spawn_npc.py --port 4000 --tm-port 4050 # TM-Client
Is this way to fix your problem ?

In my problems, The code that I want spawn 10 vehicles without sensors is in terminal2. And the code that spawn one vehicle with lidar is in terminal 3.

makaveli10, 2023-05-30:
> @ronyshaji yes... by using different traffic manager port for different clients.

Use different port for traffic manager in different clients. And that should solve the issue of vehicles going wild.