carla-simulator/carla #5187: [Carla 0.9.12] Building error in Ubuntu 18.04

tengxiansen, 2022-02-20:
When I execute "make PythonAPI", I met the error shown below(No use to re-execute this problem)
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c5-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-0b13b0-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
**fatal: bad revision 'HEAD'
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 128**

lyjlalala, 2022-03-14:
Same issue here. Have you found a solution?

noobC7, 2022-03-27:
Same issue. Sad.

zehranrgi, 2023-06-15:
is there any solution for that?

zehranrgi, 2023-06-17:
**Solution**
If you face this problem, for me, I tried to get the previous version of Carla from the releases pages by downloading it. The problem is because of the git. So, take the git folder from the latest version, paste it to your current version folder, and it should be worked. And, see if the `git branch` gives  `*master` answer.

Blyron, 2024-04-12:
Sorry for not being able to help you. Are you happy to close this issue as you have found the solution?

closed 2024-04-19 by Blyron