carla-simulator/carla #5190: Using Carla as visualizer for trajectory file (csv data)

brunnerpascal, 2022-02-21:
Hi together,

I have trajectory files (x,y position, heading and vehicleID every 100ms) and would love to visualize them with CARLA.
For the Map I have a FBX and OpenDRIVE file.
Does anybody know if there is already any functionality for that or a Github repository for that purpose? 

I would be pleased about input

closed 2023-06-02 by stale[bot]

pjk19942, 2023-09-05:
Hi, was wondering if you found a way?

jiaqi-lian, 2023-09-26:
hi, I wonder if you've found a way

pjk19942, 2023-09-26:
Hi, yeah kind of. I have a csv file with X,Y Z and Yaw Pitch and roll data along with timestamp . All i had to do was to extract the the data from the csv and then pass it as a location at every time stamp to my vehicle such as below. 

# Extracting CSV data into a variable , which includes timestamp, XYZ coordinates, and Yaw, Pitch, and Roll angles.
for timestamp, xyz, ypr in my_trajectory:
    # Creating a rotation object based on the extracted Yaw, Pitch, and Roll angles.
    rotation = carla.Rotation(yaw_csv, pitch_csv, roll_csv)  # Replace with the desired rotation values

    # Spawning a vehicle actor in the world using the chosen vehicle blueprint and setting its transformation.
    vehicle = world.spawn_actor(vehicle_blueprints[0], carla.Transform(location))
    vehicle.set_transform(carla.Transform(carla.Location(x=x_csv, y=y_csv, z=z_csv), rotation))
    time.sleep(sampletime)

jiaqi-lian, 2023-09-27:
Maybe you misunderstood me, I would rather like to know if there is a way to generate an opendrive format map from a csv file

liuyueChang, 2024-01-17:
Hi, was wondering if you found a way?

vollkornholztuer, 2024-11-27:
I also have a similar use case for carla. Have you found a solution?

minhaj6, 2025-04-11:
Have any of you found any solution 🤒