carla-simulator/carla #5191: CarlaUE4 stuck on the first screen

js515, 2022-02-21:
Hello, I'm trying to launch Carla simulation.
I tried to start the test run as below. But when I execute it, the screen comes out, and no other option is available.
Looking at examples from other people, the control buttons or other options are also available.
What can I do to go forward to use this simulator?
I very much appreciate your concern on this matter!

[image]

zm0612, 2022-03-20:
Hi, have you found the answer?

js515, 2022-03-21:
Hello, No, I haven't found the answer yet.

js515, 2022-03-21:
> Hi, have you found the answer?

Hello, No, I haven't found the answer yet.

smp4966, 2022-07-05:
hii @js515 

i think , you have to rebuild again. its seem to be like somthing insatlltion is not fix yet

smp4966, 2022-07-05:
I think  you need to rebuild again. seems to be something the install hasn't fixed yet

Blyron, 2025-08-05:
It needs to load assets, 1st time of execution takes time to load if it is in a HDD disk. Is this solving your question?