carla-simulator/carla #5193: Getting reflectivity from lidar sensor

pespen, 2022-02-22:
I am trying to use the Carla lidar sensor to create 2D images from 3D point cloud data(similarly to how Ouster does it), and have so far been successful with the range and intensity images. Ouster also creates what they call Ambient images, where they somehow use the reflectivity that they get from their sensor to create a more life-like image. Is it possible to get this information from Carla? I basically need to know the number of photons over time the sensor registers. See the image for reference

[image]

GruNyv, 2022-03-03:
Hi, sounds like we are working on pretty similar problems. However, note that the ambient images are not the same as the reflectivity image. The ambient image are created based on the photons coming from other sources than the lidar (the sun), while the reflectivity images are created by using the inverse square law for the intensity. Btw, could I ask you if you have got any realistic results for the intensity image in Carla? From my experience the simulated values are pretty unrealistic for the intensity image you get from the Ouster sensor.

pespen, 2022-03-03:
Alright, that makes sense about the reflectivity and ambient images. My images are not as good as the ones from Ouster, I'm not sure if it's because they do more processing or if they just have more data available. Here are some examples of range and intensity.
[image]
[image]

GruNyv, 2022-03-03:
Ouster does not project the point cloud from 3D space to 2D as I assume you do here to create the 2D image. Instead they use the values that each pixel in the sensor itself measures, since these are organized into an array of num_channels * width

GruNyv, 2022-03-03:
Btw, if you want we could talk together about this problem on an other platform, since I do not want to talk about all parts of my thesis in public. Could be interesting to get some other input

pespen, 2022-03-03:
Sure! Shoot me an email at phespen@stud.ntnu.no, and we can take it from there (probably use Discord or something if that is OK)

TzabarDolev, 2023-04-16:
Hi,
Can I ask how did you get the reflectivity and range images from CARLA?

Blyron, 2024-04-19:
I am closing the issue

closed 2024-04-19 by Blyron