carla-simulator/carla #5196: No rule to make target 'CarlaUE4Editor'.  Stop

Alphabaijinde, 2022-02-23:
Env Info:
ubuntu 18.04  
carla 0.9.13 
UE4.18

Compile Info:

Setting up Unreal Engine 4 project files...

+ xbuild Source/Programs/UnrealBuildTool/UnrealBuildTool.csproj /verbosity:quiet /nologo /p:TargetFrameworkVersion=v4.5 /p:Configuration=Development
+ mono Binaries/DotNET/UnrealBuildTool.exe -projectfiles -project=/home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/CarlaUE4.uproject -game -engine -makefiles
Discovering modules, targets and source code for project...
While compiling /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Intermediate/Build/BuildRules/CarlaUE4ModuleRules.dll:
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(36,9) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(37,9) : error CS0103: The name `PrivateDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(43,9) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(44,9) : error CS0103: The name `PrivateDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(135,25) : error CS1502: The best overloaded method match for `System.Collections.Generic.List<UnrealBuildTool.ModuleRules.RuntimeDependency>.Add(UnrealBuildTool.ModuleRules.RuntimeDependency)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(135,29) : error CS1503: Argument `#1' cannot convert `string' expression to type `UnrealBuildTool.ModuleRules.RuntimeDependency'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(142,25) : error CS1502: The best overloaded method match for `UnrealBuildTool.ModuleRules.RuntimeDependencyList.Add(string, UnrealBuildTool.StagedFileType)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(142,42) : error CS1503: Argument `#2' cannot convert `string' expression to type `UnrealBuildTool.StagedFileType'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(208,29) : error CS1502: The best overloaded method match for `System.Collections.Generic.List<UnrealBuildTool.ModuleRules.RuntimeDependency>.Add(UnrealBuildTool.ModuleRules.RuntimeDependency)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(208,38) : error CS1503: Argument `#1' cannot convert `string' expression to type `UnrealBuildTool.ModuleRules.RuntimeDependency'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(209,29) : error CS1502: The best overloaded method match for `System.Collections.Generic.List<UnrealBuildTool.ModuleRules.RuntimeDependency>.Add(UnrealBuildTool.ModuleRules.RuntimeDependency)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(209,38) : error CS1503: Argument `#1' cannot convert `string' expression to type `UnrealBuildTool.ModuleRules.RuntimeDependency'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(210,29) : error CS1502: The best overloaded method match for `System.Collections.Generic.List<UnrealBuildTool.ModuleRules.RuntimeDependency>.Add(UnrealBuildTool.ModuleRules.RuntimeDependency)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(210,38) : error CS1503: Argument `#1' cannot convert `string' expression to type `UnrealBuildTool.ModuleRules.RuntimeDependency'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(211,29) : error CS1502: The best overloaded method match for `System.Collections.Generic.List<UnrealBuildTool.ModuleRules.RuntimeDependency>.Add(UnrealBuildTool.ModuleRules.RuntimeDependency)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(211,38) : error CS1503: Argument `#1' cannot convert `string' expression to type `UnrealBuildTool.ModuleRules.RuntimeDependency'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(212,29) : error CS1502: The best overloaded method match for `System.Collections.Generic.List<UnrealBuildTool.ModuleRules.RuntimeDependency>.Add(UnrealBuildTool.ModuleRules.RuntimeDependency)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(212,38) : error CS1503: Argument `#1' cannot convert `string' expression to type `UnrealBuildTool.ModuleRules.RuntimeDependency'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(213,29) : error CS1502: The best overloaded method match for `System.Collections.Generic.List<UnrealBuildTool.ModuleRules.RuntimeDependency>.Add(UnrealBuildTool.ModuleRules.RuntimeDependency)' has some invalid arguments
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(213,38) : error CS1503: Argument `#1' cannot convert `string' expression to type `UnrealBuildTool.ModuleRules.RuntimeDependency'
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(229,5) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(230,5) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(231,5) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(232,5) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(233,5) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: /home/baijinde/study/3rdparty/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(234,5) : error CS0103: The name `PublicDefinitions' does not exist in the current context
ERROR: UnrealBuildTool Exception: Unable to compile source files.

rayneong, 2022-03-07:
I had a similar error b/c I was accidentally building on an older UE (probably 4.18). If you're trying to build CARLA 0.9.13, you should probably be using UE4.26 instead. Hope this helps!

Alphabaijinde, 2022-03-07:
这是来自QQ邮箱的假期自动回复邮件。您好，我已经收到您的邮件，会尽快回复的。

Alphabaijinde, 2022-03-09:
> 我有一个类似的错误 b/c 我不小心在较旧的 UE（可能是 4.18）上构建。如果您尝试构建 CARLA 0.9.13，您可能应该改用 UE4.26。希望这可以帮助！

o, thank u ! I refer to the project compilation document, which uses 4.18！ UE4.24 and CARLA0.9.13 ？

closed 2024-05-10 by Blyron