carla-simulator/carla #5209: make PythonAPI failed

yeapzexin, 2022-02-24:
Hi there,

I am using master branch which I supposed is tag 0.9.13 Carla version.
I am using Ubuntu 18.04.6.
I have been trying to run the command "make PythonAPI"  however it failed with the result shown below:

```
.
.
.
.
.
.
         ^
In file included from /home/yeap/carla/Build/rpclib-v2.2.1_c5-c8-source/lib/rpc/server.cc:1:
In file included from /home/yeap/carla/Build/rpclib-v2.2.1_c5-c8-source/include/rpc/server.h:6:
In file included from /home/yeap/carla/Build/rpclib-v2.2.1_c5-c8-source/include/rpc/config.h:7:
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:153:8: error: no member named 'int8_t' in the global namespace
using::int8_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:154:8: error: no member named 'int16_t' in the global namespace
using::int16_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:155:8: error: no member named 'int32_t' in the global namespace
using::int32_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:156:8: error: no member named 'int64_t' in the global namespace
using::int64_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:158:8: error: no member named 'uint8_t' in the global namespace
using::uint8_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:159:8: error: no member named 'uint16_t' in the global namespace
using::uint16_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:160:8: error: no member named 'uint32_t' in the global namespace
using::uint32_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:161:8: error: no member named 'uint64_t' in the global namespace
using::uint64_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:163:8: error: no member named 'int_least8_t' in the global namespace
using::int_least8_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:164:8: error: no member named 'int_least16_t' in the global namespace
using::int_least16_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:165:8: error: no member named 'int_least32_t' in the global namespace
using::int_least32_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:166:8: error: no member named 'int_least64_t' in the global namespace
using::int_least64_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:168:8: error: no member named 'uint_least8_t' in the global namespace
using::uint_least8_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:169:8: error: no member named 'uint_least16_t' in the global namespace
using::uint_least16_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:170:8: error: no member named 'uint_least32_t' in the global namespace
using::uint_least32_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:171:8: error: no member named 'uint_least64_t' in the global namespace
using::uint_least64_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:173:8: error: no member named 'int_fast8_t' in the global namespace
using::int_fast8_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:174:8: error: no member named 'int_fast16_t' in the global namespace
using::int_fast16_t;
     ~~^
/home/yeap/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:175:8: error: no member named 'int_fast32_t' in the global namespace
using::int_fast32_t;
     ~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
5 warnings and 20 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 1
```

After reading from the forum, I suspect that this is the issue due to the clang llvm . Hence, I try to delete and purge the whole clang and reinstall again following the link: https://carla.readthedocs.io/en/latest/build_linux/ . It still does not work. This is the result after I run clang --version:

```
clang version 8.0.1- (branches/release_80)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

Have any of you encountered this issue and how did you solve it? Please teach me. Thanks and appreciated.

Utopialjp, 2022-08-16:
I had the same problem, did you solve it?

Utopialjp, 2022-08-16:
On my computer, there seems to be a conflict between carla/Build/llvm-8.0-install/include/c++/v1 and /usr/include/c++/v1。
When I removed or renamed /usr/include/c++/v1, the problem was solved.
The path /usr/include/c++/v1 may be installed when I install libc++abi1 or libc++abi-dev. So uninstalling it will solve the problem.

closed 2024-05-17 by MarcelPiNacy-CVC

huangzhuofei, 2024-12-25:
I solved this problem using a method similar to @Utopialjp 
Thank you.

I searched v1 in / folder, and rename it.

which I find in my system is :
sudo find / -name v1
/usr/lib/llvm-10/include/c++/v1

then I rename it to v2, it works

BUT, I can't believe that It worked in llvm-8 but NOT in llvm-10