carla-simulator/carla #5211: make launch  on Win10 throws error:   YieldSignComponent.cpp(28): error C2027: use of undefined type 'carla::road::element::RoadInfoSignal'

JonathanSchmalhofer, 2022-02-24:
Hi everyone,

I try building CARLA from scratch on Windows because I want to import my own custom maps in CARLA built from source, as recommended [here ](https://carla.readthedocs.io/en/latest/tuto_M_custom_map_overview/)and [here](https://carla.readthedocs.io/en/latest/tuto_M_add_map_source/).

I follow [this tutorial for Windows Build](https://carla.readthedocs.io/en/latest/build_windows/).

When [building CARLA](https://carla.readthedocs.io/en/latest/build_windows/#build-carla) I do get many compiler errors on _make launch_ (see [log.txt](https://github.com/carla-simulator/carla/files/8137131/log.txt)).

CARLA version: _master (commit hash 8854804f4d7748e14d937ec763a2912823a7e5f5 )_
Platform/OS: _Windows 10 Pro 19042.1526_
Problem you have experienced: _When building CARLA myself in Windows, make launch throws compiler errors (see attached [log.txt](https://github.com/carla-simulator/carla/files/8137131/log.txt))._
What you expected to happen: _No build error and successful start of Unreal Engine._
Steps to reproduce: _Follow tutorial [Windows build](https://carla.readthedocs.io/en/latest/build_windows/#unreal-engine) until step make launch (I performed make LibCarla before make launch)._
Other information:
see attached log of command "make launch": [log.txt](https://github.com/carla-simulator/carla/files/8137131/log.txt)
_GNU Make 3.81
cmake version 3.21.4
Python 3.7.9 x64
[CarlaUnreal](https://github.com/CarlaUnreal/UnrealEngine.git) on branch carla (commit hash 68061ec34a1503dd995dbf37d28607bfef220eea )
CarlaUnreal built without problems and UE4Editor could be started_

I did not find any similar issue here using the search function (except [4878](https://github.com/carla-simulator/carla/issues/4878), but I try to build the regular CARLA build without changes). Thanks for any help.

Attached log file: [log.txt](https://github.com/carla-simulator/carla/files/8137131/log.txt)

52bloo, 2022-02-25:
Hello. I'd like to chime in to report experiencing a very similar issue. 

You said you are running into the issue with an unmodified version of Carla. That's interesting because in my case I get the exact same error messages with you ONLY when I install a customized code that builds upon Carla. I am using a 3rd-party developed library (DReyeVR) that builds upon Carla had this issue and spent a whole week trying to figure this out. For me with just a base installation of Carla `make launch` worked without issues, but when I added the library it caused the exact same issue. You can see the troubleshooting I did with the developers of the library [here](https://github.com/HARPLab/DReyeVR/issues/5): 

For me it turned out that having the custom sensor declaration code causes this behavior but not consistently; I was able to reproduce it in the two different x64 Windows devices I had, but the developers of the library were not able to do so. So the exact way to reproduce the issue is unknown. It's interesting that in your end it happens without any modifications.

There was a similar [unresolved github issue](https://github.com/carla-simulator/carla/discussions/4823#discussion-3660766) in carla, but they also had some customized sensor codes going on.

I very much hope the carla team can look into this, as I think there's a lot of potential for using customized sensor code.

I'd like to compare notes, so we get a better idea of what could be causing it. So here were the common hardware and software factors between the two machines I have that exhibited this behavior: 

Hardware

- NVidia GPU (different specific models, 10xx and 30xx)
- AMD 5xxx CPU (different specific models. I'm aware there are some software differences between Intel and AMD CPUs, such as BLAS related libraries, but I'm not sure that's relevant here.)
Software

x64 Windows 10
- system locale was at least temporarily changed to non latin alphabet based language at some point
- Other software requirements for building carla from source
- python 3.x x64 (conda/native agnostic, tried with 3.8 and 3.9)
- Carla related required environment variables (%CARLA_ROOT%, %UE4_ROOT%, cmake, GNU make etc.)
- source file encodings(both carla and DReyeVR) not altered from original

GustavoSilvera, 2022-02-26:
Hello there, I encountered this error as well and would like to suggest trying the fix provided in #5217. Looking forward to putting this error behind us.

52bloo, 2022-02-28:
I have found that the error indeed no longer arises in my case once I applied GustavoSilvera's changes mentioned in the pull request.

JonathanSchmalhofer, 2022-02-28:
Indeed, this solves the issue for `YieldSignComponent.cpp`, however I still get hundreds of other errors (as before), first error this time:

```
Building 250 actions with 8 processes...
  [1/250] TrafficSignBase.gen.cpp
  [2/250] ActorDefinition.gen.cpp
  [3/250] ActorDescription.gen.cpp
  [4/250] CarlaHUD.gen.cpp
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(53): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(54): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  [5/250] DefaultMovementComponent.cpp
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(61): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(161): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(25): warning C4996: 'AWheeledVehicle::GetVehicleMovementComponent': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(26): warning C4996: 'UWheeledVehicleMovementComponent::SetThrottleInput': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(27): warning C4996: 'UWheeledVehicleMovementComponent::SetSteeringInput': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(28): warning C4996: 'UWheeledVehicleMovementComponent::SetBrakeInput': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(29): warning C4996: 'UWheeledVehicleMovementComponent::SetHandbrakeInput': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(32): warning C4996: 'UWheeledVehicleMovementComponent::SetUseAutoGears': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(33): warning C4996: 'UWheeledVehicleMovementComponent::SetTargetGear': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(37): warning C4996: 'UWheeledVehicleMovementComponent::SetUseAutoGears': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\MovementComponents\DefaultMovementComponent.cpp(40): warning C4996: 'UWheeledVehicleMovementComponent::SetTargetGear': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer co [... cut ...]
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Walker\WalkerController.cpp(174): error C2027: use of undefined type 'ACharacter'
  D:\Work\unreal_mod\Engine\Source\Runtime\Engine\Classes\GameFramework/CharacterMovementComponent.h(24): note: see declaration of 'ACharacter'
  D:\Work\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Walker\WalkerController.cpp(178): error C2027: use of undefined type 'ACharacter'
  D:\Work\unreal_mod\Engine\Source\Runtime\Engine\Classes\GameFramework/CharacterMovementComponent.h(24): note: see declaration of 'ACharacter'
```

JonathanSchmalhofer, 2022-02-28:
As I need to progress in my current project, I switched to Ubuntu 20.04, and there the latest version (same git commit) builds with (almost) no issues.

closed 2023-06-02 by stale[bot]