carla-simulator/carla #5213: Carla crashes with  error VK_ERROR_DEVICE_LOST

wu0023, 2022-02-25:


betray12138, 2022-03-16:
Hello，could you please tell me if the problem has been solved ~

betray12138, 2022-03-16:
> _No description provided._

Hello，could you please tell me if the problem has been solved ~

wu0023, 2022-03-16:
Hi,

Not sure how the problem is solved. But the Carla environment finally worked by using the Carla gym environment and 0.9.9.4 provided by Secant work.

Link:
https://github.com/LinxiFan/SECANT

Maybe you can have a try on that.

Regards,
Wu Yuting

betray12138, 2022-03-16:
> Hi, Not sure how the problem is solved. But the Carla environment finally worked by using the Carla gym environment and 0.9.9.4 provided by Secant work. Link: https://github.com/LinxiFan/SECANT Maybe you can have a try on that. Regards, Wu Yuting
> […](#)

Ops! my current training version of Carla is back to 0.9.9.4, but the problem still exists.  Thanks for your replying!

wu0023, 2022-03-16:
My RL training also crashes after hours of running because of the same fault which confused me for days, and I believe the reason of that may be caused by the code of environment (e.g., env.reset()).

Instead of using vehicle destroy function at end of  each episode, the SECANT environment uses ego_vehicle.set_transfrom(new_location) to reset (teleport) the position of the actor, and which is the main difference between my original code.

Details in SECANT<https://github.com/LinxiFan/SECANT>/secant<https://github.com/LinxiFan/SECANT/tree/main/secant>/envs<https://github.com/LinxiFan/SECANT/tree/main/secant/envs>/carla<https://github.com/LinxiFan/SECANT/tree/main/secant/envs/carla>/adapter.py line 712 - 716

Regards,
Wu Yuting

betray12138, 2022-03-16:
> My RL training also crashes after hours of running because of the same fault which confused me for days, and I believe the reason of that may be caused by the code of environment (e.g., [...]

!!Thanks for your quick response and sincere reply, I will have a try immediately.  Additionally, whether you have successed in trainging the RL agent by the set_transform function instead of re-spawning after destroying the vehicle ? ~

wu0023, 2022-03-16:
The RL training works fine for now (at least it will not crash before 200K steps, which is my target step), and I directly modify on the code provided by SECANT work.

Regards,
Wu Yuting

betray12138, 2022-03-16:
> The RL training works fine for now (at least it will not crash before 200K steps, which is my target step), and I directly modify on the code provided by SECANT work. Regards, Wu Yuting From: [...]

when I use the set_transform function onto the vehicle, the sensor attaching on it seems not to change the location along with the vehicle :
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 232 (sensor.camera.rgb) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 233 (sensor.camera.depth) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 235 (sensor.other.collision) 

could you please tell me how can I solve this problem ? ~~

betray12138, 2022-03-17:
> The RL training works fine for now (at least it will not crash before 200K steps, which is my target step), and I directly modify on the code provided by SECANT work. Regards, Wu Yuting From: [...]

wow i'm glad to see my model have successfully finished 10000 training steps. It seems the problem has been solved. Thanks for your advice ~

wu0023, 2022-03-17:
Glad of that

closed 2024-05-17 by MarcelPiNacy-CVC