carla-simulator/carla #5221: Problem with "Generate Pedestrian Navigation" in version 0.9.13

huhaocong, 2022-03-01:
Hi,
I'm using carla 0.9.13 under ubuntu 20.04.
The map i used was created by RoadRunner(then import to carla) named "test".

According to the documenthttps://carla.readthedocs.io/en/0.9.13/tuto_M_generate_pedestrian_navigation/,  i changed the map's sidewalk's object name to Road_Sidewalk_test_01(it's old name was test_Roads_SidewalkNode).

Then i selected all with CTRL +A and clicked Carla Exporter. 

Then i copied the test.bin file to the carlacache folder, the map‘s folder's structure is like below:

-carlacache
    -0.9.13-****-dirty
        -map_package
            -Maps
                -test
                   -OpenDrive
                       -test.xodr
                   -TM
                       -test.bin

But when i execute the script file : generate_traffic.py, an error occured like below：

****************************************************************************************************************************
emggo@emggo-Precision-3640-Tower:~/carla-0.9.13/carla/PythonAPI/examples$ python3 generate_traffic.py 
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: Traffic sign 98 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal. 
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: Traffic sign 98 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal. 

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "generate_traffic.py", line 379, in <module>
    main()
  File "generate_traffic.py", line 168, in main
    traffic_manager = client.get_trafficmanager(args.tm_port)
RuntimeError: s <= road->GetLength()
emggo@emggo-Precision-3640-Tower:~/carla-0.9.13/carla/PythonAPI/examples$ 
****************************************************************************************************************************

And i tired make a folder(Nav) under Maps and putted the test.bin file in it, and the error also occured!!

Why that happened?

Glad about any help!

Thanks！！

SergioBarroso, 2022-05-17:
We have same problem here:

https://github.com/carla-simulator/carla/issues/5449

huhaocong, 2022-06-14:
已解决！
(1)检查地图资产列表中所有的Sidewalk、Crosswalk格式是否正确，正确格式为：Road_Sidewalk_***或者Road_Crosswalk_***
(2)检查自己手动添加的Plane（或者SidewalkPlane）是否在物理上联通了两条便道，即手动添加的Plane两端必须覆盖上两侧便道的路面
(3)生成的.bin文件必须放置到地图文件夹内部的与OpenDrive文件夹的同一目录层级下的Nav目录中（如果没有，自己创建一个文件夹），目录结构示例：
Content/YourMapFodler/Maps/YourMap/Nav/***.bin
(4)运行之前，删除Carla的缓存文件（如果没有就不用管它了），一般缓存文件夹名称为CarlaCache，Linux系统下在你的home目录里（Windows系统的不知道在哪），找到你的地图的对应文件夹，直接全部删除就可以了

closed 2022-06-14 by huhaocong

yangbiaoLeslie, 2022-08-04:
> 已解决！ (1)检查地图资产列表中所有的Sidewalk、Crosswalk格式是否正确，正确格式为：Road_Sidewalk__**或者Road_Crosswalk_**_ (2)检查自己手动添加的Plane（或者SidewalkPlane）是否在物理上联通了两条便道，即手动添加的Plane两端必须覆盖上两侧便道的路面 [...]

可以交流下行人如何生成吗？我这边都符合，但是就是不能生成
vx或者邮箱18059866261@163.com可以沟通下吗？谢谢啦

donghaiwang, 2023-12-25:
> 已解决！ (1)检查地图资产列表中所有的Sidewalk、Crosswalk格式是否正确，正确格式为：Road_Sidewalk__**或者Road_Crosswalk_**_ (2)检查自己手动添加的Plane（或者SidewalkPlane）是否在物理上联通了两条便道，即手动添加的Plane两端必须覆盖上两侧便道的路面 [...]

windows是在 C:\Users\Administrator\carlaCache 目录下，.bin文件具体在：C:\Users\Administrator\carlaCache\0.9.14-4-gf14acb257-dirty\Carla\RoadRunner\Maps\Nav\HutbCarlaCity.bin