carla-simulator/carla #5230: Possible documentation error of recorder file

dvdhfnr, 2022-03-04:
CARLA version: 0.9.13
Platform/OS: Linux
Problem you have experienced: Documentation does not fit to code.
What you expected to happen: When recording the `yaw` angle in the recorded vehicle position should change. However, it stays close to 0, and the `roll` is changing; if interpreting the last element of the rotation as `roll` - according to the [docu](https://carla.readthedocs.io/en/latest/ref_recorder_binary_file_format/#packet-6-position)
Steps to reproduce: Record a file while a certain car is performing a turn, and check the binary recorder file.
Other information (documentation you consulted, workarounds you tried): https://carla.readthedocs.io/en/latest/ref_recorder_binary_file_format/#packet-6-position

I suppose that the actual rotation angle order in the recorder file is `(roll, pitch, yaw)` instead of `(pitch, yaw, roll)`. According to https://github.com/carla-simulator/carla/blob/8854804f4d7748e14d937ec763a2912823a7e5f5/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp#L158 the recorder does not write a Carla rotation object (which would be `(pitch, yaw, roll)`) to the recorder file, but the Euler angles of a `FTransform`. Here, the order might be `(roll, pitch, yaw)` - not 100% sure about this, but `yaw` seems to be at the last position. This is the entry that changes when a car is performing a turn.

It seems to be only an error in the documentation of the recorder file (https://carla.readthedocs.io/en/latest/ref_recorder_binary_file_format/#packet-6-position). Since the data is written and read in the same ways, there is no "bug" in the code, and everything seems to work fine.

Thanks for looking into this!

Blyron, 2024-05-10:
You are right. We will be fixing this soon

M-Colley, 2026-10-24:
The position-packet documentation now gives the rotation order as **roll, pitch, yaw**, in degrees: [Packet 6 — Position](https://github.com/carla-simulator/carla/blob/defabe35d1ad47c4231d4f7fb1f4e46ac2e08ec5/Docs/ref_recorder_binary_file_format.md#L238), checked on `ue58-dev` at `defabe35d1ad47c4231d4f7fb1f4e46ac2e08ec5`.

The [recorder writer](https://github.com/carla-simulator/carla/blob/defabe35d1ad47c4231d4f7fb1f4e46ac2e08ec5/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp#L182) still records `Transform.GetRotation().Euler()`. The documentation-order correction requested here is present. This was a documentation/source check; I did not generate or inspect a new binary recording.

closed 2026-11-12 by Madecu