carla-simulator/carla #5232: Where in source is the autopilot functionality specified in 0.9.13?

lttnml1, 2022-03-04:
I'm trying to take some code I used from an old research project in CARLA 0.9.6 to 0.9.13.  I'm having issues compiling the source for 0.9.6 with UE 4.22, and I want to make use of some additional functionality in 0.9.13.

In 0.9.6, /Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.cpp used to have a function called TickAutopilotController which contained much of the logic, as well as GoToNextTargetLocation, CalcStreeringValue, etc.  I don't see these in 0.9.13 - where does this code currently live?

jackbart94, 2022-03-09:
Since 0.9.7 the autopilot has been deprecated, and the Traffic Manager replaced it. It's definitely not available in 0.9.13.

lttnml1, 2022-05-05:
Fantastic - thanks for your help.

closed 2022-05-05 by lttnml1