carla-simulator/carla #5240: problem related to 'previous_until_lane_start' method

Spritepang, 2022-03-09:
CARLA version:0.9.12
Platform/OS: Ubuntu `20.04`
Problem you have experienced: 
  I am trying to get point info from python API in Town03 and here is part of my code :
```
client = carla.Client('localhost',2000)
client.set_timeout(10)
world = client.get_world()
map = world.get_map()    
test_waypoint = map.get_waypoint_xodr(65,2,0.1)                    # 1173 ,5 
test_list = test_waypoint.previous_until_lane_start(0.5)
for index,point in enumerate(test_list):
    world.debug.draw_point(point.transform.location, size=0.05, color = carla.Color(255,0,0), life_time=60)
    print('point %d  location is %r,%r, road id is %r' %(index,point.transform.location.x,point.transform.location.y,point.road_id))
```
I select the  waypoint locating at road 65 ,lane 2 and use  previous_until_lane_start to get the full waypoint list, and I print their location and road_id. It turns out that the last point does not belong to this road and lane, the pic below shows the result.
[image]
[image]
You can reproduce this problem using the code above. In my experiment there are more than one lanes that have the same issue.
please let me know if it is a bug or if I somehow misunderstand and wrongly use this method , thanks !

renaudponcelet, 2023-02-06:
I have the same issue (on Carla 9.14 on Windows). I think it's an issue from the method, for me (at least on Town01) all lanes have this extra waypoint. If it's an issue with the method, is it safe to always remove the last waypoint, or do we have to filter with their road id ? (which makes the method useless…)

renaudponcelet, 2023-02-06:
Ok, I think it is an issue with the method. The method adds previous waypoints with the target length until the previous waypoint is no more in the lane. But then the last waypoint is not really the lane start waypoint. The distance remaining is computed either from `s` if the `road_id` is positive or from `road_length - s` if `road_id` is negative. And then a last waypoint is computed with the previous method and this remaining distance. And this doesn't work in the C coded method because I manage to replicate in python, and it's working. Something like that does the trick :

```
waypoints = waypoint.previous_until_lane_start(0.5)
# I remove the bugging lane start
waypoints.pop(-1)
# I compute my own lane start
lane_start = waypoints[-1].previous(waypoints[-1].s)
# I append it to have the full lane
waypoints.append(lane_start)
```

Regarding of the python hack and the extra waypoint it seems to me that `road_length - s` is wrongly used and, in our case, is almost equal to `road_length` so we have a random waypoint far before the lane start.

Sorry, I don't propose a fix of the method itself, but I don't have the environment to test it…

M-Colley, 2026-10-26:
Draft PR #10016 proposes the `ue58-dev` fix for the backward-distance calculation in `previous_until_lane_start`, with native regressions for road endpoints and both traffic directions. Town03/live-simulator validation remains outstanding; the UE4 branch is outside that PR.