carla-simulator/carla #5244: Error compiling server, carla 0.9.13 Ubuntu 20.4

FernandoGarImb, 2022-03-10:
[image]
When I do "make launch" this error appears in the terminal, I don't know if it is a python error or if I am missing a package to install.
Thanks in advance for your help.

jayanshuagrawal, 2022-08-18:
Hi 

Did you got some solution for this issue, I am also facing this issue while trying to do make launch 

Thanks in advance

jayanshuagrawal, 2022-08-18:
Hi 
Found the solution for this problem, I noticed that Python3 was installed. so I used the below command to link python and python3 

**sudo apt install python-is-python3**

It worked for me. Hope it will work for others too

Blyron, 2024-04-19:
Are you happy to close this issue as you have found the solution?

closed 2024-05-03 by Blyron