carla-simulator/carla #5247: How to custom path for non-ego vehicle?

L1nmmm, 2022-03-11:
Hello,

Currently I have trajectory data, which contains the x,y,z coordinates, speed, etc. of the vehicle every 100 ms. Is it possible for Carla to make the NPC vehicle drive along a fixed trajectory? I've tried using the Traffic Manager.set_path function from version 0.9.13 and it didn't do what I wanted. Or can I set up a non-ego vehicle that moves along a customized path? 

Thank you!

werewolfdev, 2022-03-14:
You can provide custom paths to non ego vehicles by defining scenarios using openSCENARIO scripts in scenario runner . There is a routing action class

L1nmmm, 2022-03-17:
> You can provide custom paths to non ego vehicles by defining scenarios using openSCENARIO scripts in scenario runner . There is a routing action class

Thank you for your help!

Do you mean the "RouteScenario Class" in "route_scenario. py"? I checked this class and found that it uses "waypoints" to describe the path of vehicle movement like other methods. However, if I set the destination in a position where there are no waypoints nearby, the vehicle cannot reach the destination as well.
Did I get it wrong, or is there another way?

werewolfdev, 2022-03-17:
I actually mentioned regarding openSCENARIO based scenario generation. I havent actually used route_scenario.py. You can find examples of openSCENARIO scripts (xosc) in the examples folder inside scenario runner . Scenario runner has to be seperately downloaded from github according to the version of CARLA you are using . In the ASAM openSCENARIO website you can find the mentioned action RoutingAction

closed 2025-04-01 by Blyron