carla-simulator/carla #5257: Errors of import map from Roadrunner containing traffic lights?

huhaocong1, 2022-03-15:
Hi, I'm having some issues importing map generated from Roadrunner. 

The environment of my computer is :
Ubuntu-20.04
UE-4.26.2
Calra-0.9.13
RoadRunner2021b

I created one test map,contains lot of props of RoadRunner.After i imported the map (use "make import")to carla,in the edit mode,i could not seen any traffic_light or signal_post(in RoadRunner i putted all kinds of traffic lights and signal posts)???But i could seen other props i putted in the map.When i "play" the map,i saw two traffic lights(only two,and it was hand-up by a signal-post),why??

What should i do?

Hopes for your help!!

Thanks!

RocketDK, 2022-04-19:
Similar problem on my side with Carla 0.9.13. No traffic signs are shown in CARLA after importing a map from RoadRunner. I also tried to create a minimal map with only a small road and a traffic sign, and still no luck. Also: `make import` doesn't raise any errors.

huhaocong, 2022-06-14:
这个问题一直没有解决，不知道是不是carla官方与UE4之间的“协商”问题？
我后来直接使用RoadRunner Plugin方式导入地图（为了使用RoadRunner里的那些便利的地图功能而妥协了）！！
或者，只在RoadRunner内编辑路面功能（包括道路、便道、地面），然后其他所有的编辑工作（绿植、建筑、标牌等）直接转移到carla内部制作！

xiaoyang666, 2022-07-19:
> 

这个问题也困扰了我很久，用插件还是不能解决这个问题

heesungsung, 2022-08-03:
Did you solve the problem?
I have the same issue.  Traffic lights and poles do not appear in the Carla.

jannine92, 2023-02-17:
Are there any updates from anyone? I'm facing the same issue with RoadRunner R2022a and Carla 0.9.13 and I need to import the maps by using the `make import` way.

i14d4, 2023-02-28:
> Similar problem on my side with Carla 0.9.13. No traffic signs are shown in CARLA after importing a map from RoadRunner. I also tried to create a minimal map with only a small road and a traffic sign, and still no luck. Also: `make import` doesn't raise any errors.

I have the same issue as already described. Does anyone has a solution for how a map from roadrunner can be import including traffic lights?
kind regards and thanks a lot in advance !

goodgoodliving, 2023-03-27:
Yes, I've also encountered this problem. Similar to street lights, these poles don't appear on the map
My versions are Roadrunner 2022b and carla 0.9.13 .and I use the **make import** way.

goodgoodliving, 2023-03-27:
> 这个问题一直没有解决，不知道是不是carla官方与UE4之间的“协商”问题？ 我后来直接使用RoadRunner Plugin方式导入地图（为了使用RoadRunner里的那些便利的地图功能而妥协了）！！ 或者，只在RoadRunner内编辑路面功能（包括道路、便道、地面），然后其他所有的编辑工作（绿植、建筑、标牌等）直接转移到carla内部制作！

我想问一下这些编辑工作如何在carla内部制作啊？？？

goodgoodliving, 2023-03-27:
@huhaocong 我想问一下这些编辑工作如何在carla内部制作啊？？？

goodgoodliving, 2023-03-27:
> 这个问题一直没有解决，不知道是不是carla官方与UE4之间的“协商”问题？ 我后来直接使用RoadRunner Plugin方式导入地图（为了使用RoadRunner里的那些便利的地图功能而妥协了）！！ 或者，只在RoadRunner内编辑路面功能（包括道路、便道、地面），然后其他所有的编辑工作（绿植、建筑、标牌等）直接转移到carla内部制作！

@huhaocong 我想问一下这些编辑工作如何在carla内部制作啊？？？

goodgoodliving, 2023-03-27:
> Hi, I'm having some issues importing map generated from Roadrunner.
>
> The environment of my computer is : Ubuntu-20.04 UE-4.26.2 Calra-0.9.13 RoadRunner2021b
>
> I created one test map,contains lot of [...]

@huhaocong1 请问你解决了吗？？

huhaocong1, 2023-03-27:
打开ue4，找到地图，编辑，拖拽

goodgoodliving, 2023-03-27:
> 打开ue4，找到地图，编辑，拖拽
> […](#)

@huhaocong1    好的感谢！！！     roadrunner导入的方法还是没有成功是吧？

huhaocong1, 2023-03-27:
不知道啊，我早就离职了，不做那个了

  
打开ue4，找到地图，编辑，拖拽
 …

goodgoodliving, 2023-03-27:
> Did you solve the problem? I have the same issue. Traffic lights and poles do not appear in the Carla.

@heesungsung  Did you solve the problem?

goodgoodliving, 2023-03-27:
> Are there any updates from anyone? I'm facing the same issue with RoadRunner R2022a and Carla 0.9.13 and I need to import the maps by using the `make import` way.

 @jannine92  Did you solve the problem?

jannine92, 2023-04-04:
No, I still have the same problem.

sainttelant, 2025-04-01:
any updated for this issue?