carla-simulator/carla #5260: Wheel parameters cannot be changed neither via apply_physics_control() nor in UE editor

redbananamoon, 2022-03-15:
CARLA 0.9.13 compiled from source on Windows 10, running in synchronous mode.

API carla.Vehicle apply_physics_control() does not work for wheels. 
Setting any wheel parameters (such as the wheel radius) via this function has no effect.
This is also verified by calling also get_physics_control() which returns always the default values.

In addition to that, CARLA crashes after spawning an ego vehicle whose wheel radius has been modified in UE editor.

MatthiasWildeBusiness, 2022-05-25:
Hi, 
for me it worked to create new wheel object, modified here the parameter like this:

```
physics_control = vehicle.get_physics_control()
wheels = []
for i in range(4):
    wheel = physics_control.wheels[i]
    wheel_new = carla.WheelPhysicsControl(
        tire_friction = wheel.tire_friction, 
        your change
    ....
    wheels.append(wheel_new)
physics_control.wheels = wheels 
apply_physics_control(physics_control)
```

adriendod, 2023-01-28:
I've got the same problem, UE is crashing whem trying to spawn a vehicule after modifying the wheel radius... it's so weird... Anybody knows what happens?