carla-simulator/carla #5261: Carla0.9.9.4Crash--RL Train

betray12138, 2022-03-16:
Hello, I'm experiencing the terrible crash. The details are as follow.
CARLA version: 0.9.9.4
OS: ubuntu 20.04

The ERROR information on the terminal:
LowLevelFatalError [File:Unknown] [Line: 772] 
VulkanRHI::vkQueueSubmit(Queue, 1, &SubmitInfo, Fence->GetHandle()) failed, VkResult=-4
 at /home/jenkins/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanQueue.cpp:71 
 with error VK_ERROR_DEVICE_LOST
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=147680 LargeMemoryPoolOffset=278816 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

I have tried to use -opengl args to open the CarlaUE4.sh but it's doesn't make sense

The aim I used the carla simulator is to train the RL agent, but the train process is always interrupted by this awful crash!!
Please help~

ps: I would like to know using simutaneous mode to train the RL agent is feasible

[image]

kk2491, 2022-03-23:
I have the same problem, kindly let me know if you are able to fix the issue.

Thank you

betray12138, 2022-03-23:
> I have the same problem, kindly let me know if you are able to fix the issue.
> 
> Thank you

Hi~ you can use the lower version of Carla like 0.9.9.4 and use -opengl to set the rendering engine taking place the VK
Best Wishes

MarcelPiNacy-CVC, 2024-05-17:
Hi, and sorry for the long delay.

Are you still having this issue? If so, could you share what nvidia driver version you're using? Also provide the output of `nvidia-smi`.

marcel

closed 2024-09-13 by MarcelPiNacy-CVC