carla-simulator/carla #5271: Wheel rotation during playback

enricoschroeder, 2022-03-18:
Hi!

Im using the latest Carla release and have the following issue: I need to access vehicle animation data such as wheel rotation or steering angle during playback of a previously recorded scene. 

I noticed that during playback the vehicle is not animated and thus there is no valid data available for the wheel rotation. One way to remedy this is to enable physics simulation for the vehicle actor (```ego_vehicle.set_simulate_physics(True)```). This enables animation of the vehicle, however then the path of the vehicle does not match the recorded trajectory anymore (it might do so in Unreal Editor viewport, but not for exported sensor data and even in the Unreal Editor, the vehicle sometimes erraticaly jumps around - this might or might not be a different issue?)

So my question is this: How to access wheel rotation and angle of a vehicle during playback of a recorded scene?

AOOOOOA, 2023-03-24:
> Hi!
>
> Im using the latest Carla release and have the following issue: I need to access vehicle animation data such as wheel rotation or steering angle during playback of a previously recorded scene. [...]

Hi have you fixed that? If so, may I ask the exact method? Thanks!

enricoschroeder, 2023-03-24:
I think I did not find a proper solution for this. However, I am currently not working with Carla anymore, so I did not try more recent versions of it.

AOOOOOA, 2023-03-26:
> I think I did not find a proper solution for this. However, I am currently not working with Carla anymore, so I did not try more recent versions of it.

Thank you for your reply!

closed 2024-07-05 by Blyron