carla-simulator/carla #5276: Windows 11 make PythonAPI false success message possibly due to strange behavior from install_proj.bat and install_sqlite3.bat

rZapp11, 2022-03-21:
Hello everyone, I have spent the better part of spring break working on getting Carla up and running, but to no avail. I am honestly out of options and I would appreciate any help at all. A solution to the problem, ideas on what might be causing the problem, or even how to reset clean so I can start again would be greatly appreciated. I have outlined every step I took in chronological order in the "steps to reproduce section". I posted this here becasue I have seen a lot of issues with make PythonAPI, but none that have to do with proj and sclite3 (which is what I suspect to be the issue here, but I do not know enough to confirm this as I do not have a heavy CS background). Note that every time I open CMD, I open it as admin.

**CARLA version**: 0.9.13
**Python version**: 3.8.10 (Installed in A:\Program Files) (It is added to path)
**Python Modules**: numpy, pygame, setuptools, wheel, pillow, protobuf, matplotlib, pandas
**Pip version**: 21.1.1 (is within Python38 and is therefore added to path intrinsically)
**make version**: 3.81 (Installed in A:\Program Files (x86)\) (Installed with winget) (It is added to path)
**CMake version**: 3.20.21032501-MSVC_2 (Installed in A:\Program Files\) (It is added to path)
**7-Zip**: (Installed in A:\Program Files\) (It is added to path)
**git version**: git version 2.35.1.windows.2 (Installed in A:\Program Files\) (It is added to path) (Note that when in setup, I chose the default IDE as atom...not sure if that is worth mentioning)
**Visual Studio Version**: Visual Studio 2019 Community (Installed in C:\Program Files (x86)\Microsoft Visual Studio) (It is the only version of Visual Studio I have)
**SDK version**: I have Windows 11 SDK, but I also separately installed Windows 8.1 SDK after I installed VS and added it to the path because the instructions called for Windows 8 SDK (However, I do not know if it is being used at all)
**Platform/OS**: Windows 11 Pro
**Assets command**: I ran Update.bat as the assets command. Then I went to the extract location and manually extracted all the elements from the .tar.gz file
**Unreal Engine Version**: 4.26.2 (Unreal Engine is installed in C:\Program Files\) (C:\Program Files\UnrealEngine is the path I have set for the system variable UE4_ROOT)
**Problem you have experienced**: make PythonAPI reads success and says that the dist folder is created, but there is no dist folder in carla\PythonAPI\carla (I think it might have something to do with the way install_proj.bat and install_sqlite3.bat are installing into the carla\Build folder)
**What you expected to happen**: successful run of make `PythonAPI` with dist folder created under C:\Program Files\carla\PythonAPI\carla\
**Steps to reproduce**: 
* First, I installed 7-Zip, CMake, Git, and Python310 (Python38 was something that came later and I will explain soon) into A:\Program Files.
* Second, I installed the necessary Python modules, in addition to some extra ones (All of these modules are cataloged in the Python Modules section above)
* Then, I installed make 3.81 by opening cmd as admin and typing "winget install gnuwin32.make" in A:\Program Files (x86) directory
* I then Installed VS 2019 with all of the workloads and installation details mentioned in the carla install for windows instructions
* I cloned unreal engine using git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git in C:\Windows\system32
* This led to an issue when running Setup.bat, so I moved Unreal Engine to C:\Program Files
* I was then able to run Setup.bat and GenerateProjectFiles.bat successfully
* I then opened the UE4.sln file inside the source folder with Visual Studio 2019
* I selected 'Development Editor', 'Win64', and 'UnrealBuildTool'
* In the solution explorer, I right-clicked UE4 and selected Build
* Then I set my startup project to UE4
* Then I right-clicked UE4, hovered over debug, and selected start new instance
* That worked, but just to make sure everything was working properly, I launched the executable Engine\Binaries\Win64\UE4Editor.exe
* Then, I proceeded to successfully complete all of the checks outlined in the instructions to make sure that everything was installed correctly
* Then, I restarted my computer
* Then, I cloned carla using git clone https://github.com/carla-simulator/carla into C:\Windows\system32
* I then ran Update.bat which put the file in the right location, but gave me a red message output about extracting the file, so I just went to carla/Unreal/CarlaUE4/Content/Carla and extracted all of the files from the .tar.gz file myself using 7-zip (before this step, I did not have 7-zip added to the path, so immediately after this was where I added 7-zip to the path under A:\Program Files\7-Zip)
* The extracted files are shown below (note that I left the .tar.gz and .tar file there just in case):
[image]

* I then set the unreal engine environment variable by creating a system variable named UE4_ROOT that had the path of C:\Program Files\UnrealEngine
* Then I opened the x64 Native Tools Command Prompt for VS 2019 as administrator and ran make PythonAPI, which did nothing. I realized that I probably needed to move carla out of system32 and into a place closer to C: that was less restricted
* I moved carla to C:\Program Files and ran 'make PythonAPI' again, and it started doing what it was supposed to do. Although it ran, it failed with the message "Error: could not load cache" (The output from CMD is shown below):

**Beginning of code segment:**

 ```
C:\Program Files\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  32
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Program Files\carla\Build\"
-[Setup]: Creating "C:\Program Files\carla\Build\" folder...
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_zlib]: Retrieving zlib.
    -[install_zlib]: Extracting zlib from "zlib-1.2.11.zip".
    -[install_zlib]: Removing "zlib-1.2.11.zip"
    -[install_zlib]: Creating "C:\Program Files\carla\Build\zlib-source\build"
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is MSVC 19.29.30141.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 283 lines cut ...]
  client.cc
  this_handler.cc
  this_session.cc
  this_server.cc
  rpc_error.cc
  server_session.cc
  response.cc
  client_error.cc
  optional.cc
  format.cc
  posix.cc
  rpc.vcxproj -> C:\Program Files\carla\Build\rpclib-src\Release\rpc.lib
  Building Custom Rule C:/Program Files/carla/Build/rpclib-src/CMakeLists.txt
  -- Install configuration: "Release"
  -- Installing: C:/Program Files/carla/Build/rpclib-install/lib/rpc.lib
  -- Installing: C:/Program Files/carla/Build/rpclib-install/include
  -- Installing: C:/Program Files/carla/Build/rpclib-install/include/rpc
  **It proceeded to install many files in rpclib-install, but I did not include them here because as more code was thrown into this document, the document got slower and slower. If you want to see the entire output from this portion, just let me know and I will provide it**
'''

**End of code segment**
* The image below shows the Build folder after running these commands:
[image]

* **I have a few more steps I need to outline, but to make sure I do not hit the maximum number of characters, I will write them as comments under my main post**

rZapp11, 2022-03-21:
* After I saw that the rpclib-install folder was successfully created, I continued following the instructions from the PDF and ran `make PythonAPI` again
* This time, I did get a false success that told me the `dist` folder had been successfully created in` carla/PythonAPI/carla`, just like the PDF said would happen
* The code below displays the output from the command line when I ran `make PythonAPI` for the second time (Note that I wrote the code to a text file, so some of the prompts like "the syntax of the command is incorrect" were printed in CMD rather than the text file. I have done my best to manually put them in, but I definitely missed some):

**Begin code segment:**

```
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  32
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Program Files\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Program Files\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Program Files\carla\Build\" --zlib-install-dir "C:\Program Files\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Program Files\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Program Files\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_gtest]: Cloning Google Test - version "release-1.8.1"...
    -[install_gtest]: Creating "C:\Program Files\carla\Build\gtest-src\build\"
    -[install_gtest]: Generating build...
-- The C compiler identification is MSVC 19.29.30141.0
-- The CXX compiler identification is MSVC 19.29.30141.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: A:/Program Files/Python38/python.exe (found version "3.8.10") 
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Program Files/carla/Build/gtest-src
    -[install_gtest]: Building...

    -[install_gtest]: Google Test has been successfully installed in "C:\Program Files\carla\Build\gtest-install\"!
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_recast]: Cloning "Recast & Detour"
HEAD is now at 0b13b0d Adding more message details when building data, and little optimization
    -[install_recast]: Creating "C:\Program Files\carla\Build\recast-0b13b0-src\build\"
    -[install_recast]: Generating build...
-- The C compiler identification is MSVC 19.29.30141.0
-- The CXX compiler identification is MSVC 19.29.30141.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
[... 1242 lines cut ...]
running build_py
creating build
creating build\lib.win-amd64-3.8
creating build\lib.win-amd64-3.8\carla
copying source\carla\command.py -> build\lib.win-amd64-3.8\carla
copying source\carla\__init__.py -> build\lib.win-amd64-3.8\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-3.8
creating build\temp.win-amd64-3.8\Release
creating build\temp.win-amd64-3.8\Release\source
creating build\temp.win-amd64-3.8\Release\source\libcarla
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include "-IA:\Program Files\Python38\include" "-IA:\Program Files\Python38\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\includ [... cut ...]
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Program Files\carla\PythonAPI\carla\dist"!
```
**End of code section**

**If you are interested in the full output for the section above, feel free to ask and I will provide it**

**I have about 2 more steps that I took in this process, but this document is getting big and a bit slow, so I will follow this up with a comment that continues my process**
-->

rZapp11, 2022-03-21:
* At this point, no new folders have been created in Build since the last image I posted. However, when looking at the output of cmd, I noticed that it said a folder named gtest-install had been created in Build
* However, I looked in build and only gtest-src had been created
* Although this situation was never referenced on the PDF, I decided to apply the techniques I learned there to gtest to see if I could generate the gtest-install folder
* I opened up the x64 VS cmd as admin and went to C:/Program Files/carla/Build/gtest-src/build
* Then, I ran `cmake .. -G "Visual Studio 16 2019" -A x64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS_RELEASE="/MD /MP" -DCMAKE_INSTALL_PREFIX="C:/Program Files/carla/Build/gtest-install/" -DCMAKE_CXX_FLAGS=/D_SILENCE_TR1_NAMESPACE_DEPRECATION_WARNING "C:\Program Files\carla\Build\gtest-src\"`
* Once that finished running, I typed `cd ..` to go back to `C:/Program Files/carla/Build/gtest-src` and ran `cmake --build . --config Release --target install`
* After that succeeded, I checked build and saw that I now had the gtest-install folder there
* From here, I went to the carla root and ran `make PythonAPI` which gives me the following output:

**Beginning of code section:**

```
C:\Program Files\carla>make PythonAPI                                                                                                                                                                                                                                                                                        -[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all                                                                                                                                                                                                                                                                    -[Setup]: Asynchronous jobs:  32
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\Program Files\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Program Files\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Program Files\carla\Build\" --zlib-install-dir "C:\Program Files\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Program Files\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Program Files\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Program Files\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Program Files\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Program Files\carla\Build\" --toolset msvc-14.2 --version 1.72.0 -j 32
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\Program Files\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\Program Files\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
The syntax of the command is incorrect.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\Program Files\carla\Build\"
    -[install_sqlite3]: Retrieving sqlite-amalgamation-3340100.
    -[install_sqlite3]: Extracting Sqlite3 from "sqlite-amalgamation-3340100.zip".
ExpandArchiveHelper : Failed to create file 'C:\Program Files\carla\Build\sqlite-amalgamation-3340100\sqlite3.c' while expanding the archive file 'C:\Program Files\carla\Build\sqlite-amalgamation-3340100.zip' contents as the file 'C:\Program Files\carla\Build\sqlite-amalgamation-3340100\sqlite3.c' already exists.
Use the -Force parameter if you want to overwrite the existing directory 'C:\Program Files\carla\Build\sqlite-amalgamation-3340100\sqlite3.c' contents when expanding the archive file.
At C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules\Microsoft.PowerShell.Archive\Microsoft.PowerShell.Archive.psm1:397 char:17
+ ...             ExpandArchiveHelper $resolvedSourcePaths $resolvedDestina ...
[... cut ...]
* Shown below is what is inside of the proj-7.2.1 folder:
[image]
* Shown below is what is inside the sqlite-amalgamation-3340100 folder
[image]

* Every combination of methods I use seems to fail. If anyone knows how to solve the issue, I would greatly appreciate any assistance. Even if you are unsure of what to do, I would even appreciate tips on how to clean out all of the things I have done thus far so that I can start from either square 1, or the step where I clone the carla repository. Thank you for taking the time to read this message

ryunosuke1031, 2022-05-27:
I have the same problem.
Have you solved this problem?
If so, I would appreciate it if you could tell me how to proceed.