carla-simulator/carla #5278: Visualization is too bright, and blue

Leviathan321, 2022-03-22:
CARLA version: 0.9.12
Platform/OS: Ubuntu
Problem I  have experienced: Visualization is blinding, everything looks light blueish
What I expected to happen: Visiulatization with normal colours

Other information: I have tried different rendering qualities, different scenarios, but the strange colouring is still there.

TB5zhh, 2022-03-27:
I have the same issue :(
Did you find any solution?

TB5zhh, 2022-03-27:
found the cause: turns out my GPU driver fails after I turn on Secure Boot. Reinstalling `cuda-drivers` solves my problem

Leviathan321, 2022-03-27:
Ok but I want to run it on my laptop without a dedicated GPU, or is this not possible with the current version?

likith3611, 2022-08-11:
Hii even i am facing the same issue any solution to this? I installed the latest version of carla 0.9.13

xqgex, 2022-10-16:
I gave the exact same problem (Ubuntu 20.04 CARLA 0.9.13). I've tried to reinstall Cuda from [Nvidia website](https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu) but that didn't solve the problem.
[image]

Hesta256, 2022-11-17:
I have the exact same problem with Ubuntu 20.04 & CARLA 0.9.13. Reinstalling cuda drivers did not work. Any ideas?

bapukokare, 2023-07-26:
Adjust env parameter 
python ../util/environment.py -rss 1.0 --sun night --fogdist 10000 -si 0 -lg all -ds 0.0 --cloud 0.0 --weather clear  -mss 0.0 --puddles 101.0 -wet 1.0

Blyron, 2024-04-12:
You can run ./Carla.sh with -graphicsadapter=1 and it will use the dedicated Graphics card.
As you have found a solution are you happy to close this issue?

Blyron, 2024-04-19:
Since no response,
I am closing the issue, feel free to reopen

closed 2024-04-26 by Blyron

Leviathan321, 2024-04-26:
Sy for my late reply, I could resolve the issue, but I dont know any more how.