carla-simulator/carla #5280: build error (make PythonAPI)

k7m7s10, 2022-03-22:
Windows 11
Visual Studio 2019

When I build a PythonAPI(make PythonAPI) in x64 Native Tools Command Prompt for VS 2019,
error is occured. 

Please somebody help me. 

----------------------------------------------------------------------------------------------
C:\test\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\test\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\test\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\test\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\test\carla\Build\" --zlib-install-dir "C:\test\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\test\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\test\carla\Build\"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called "rpclib-src".
    -[install_rpclib]: Generating build...
-- Configuring done
-- Generating done
-- **Build files have been written to: C:/test/carla/Build/rpclib-src
    -[install_rpclib]: Building...
Error: could not load cache

    -[install_rpclib]: [Visual Studio 16 2019 Win64 ERROR] An error ocurred while installing using Visual Studio 16 2019 Win64.
    -[install_rpclib]: [Visual Studio 16 2019 Win64 ERROR] Possible causes:
    -[install_rpclib]:                - Make sure you have Visual Studio installed.
    -[install_rpclib]:                - Make sure you have the "x64 Visual C++ Toolset" in your path.
    -[install_rpclib]:                  For example using the "Visual Studio x64 Native Tools Command Prompt",
    -[install_rpclib]:                  or the "vcvarsall.bat".
    -[install_rpclib]: Exiting with error...**

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 
----------------------------------------------------------------------------------------------

dtsel, 2022-08-25:
Did you manage to resolve this? I am having the same problem. I am working on Windows Server 2022.

emlachapelle, 2023-06-21:
Hello, I am having the same error message pop up when I use the cmd make PythonAPI. 
I have:
Windows 10, python 3.8, admin x64 Native Tool Cmd Promt for VS2019 (VS 2019 plus all packages listed in instruction doc for windows build), version 3.81 of make etc. 

I feel like I have tried everything... I was even on discord trying to find a solution and got nothing.... any help would be greatly appreciated :) I have started from scratch many times at this point.