carla-simulator/carla #5283: Working with logs with a client in a container

SiddharthIVEX, 2022-03-23:
CARLA version: 0.9.12
Platform/OS: Windows + Local Ubuntu Container for client
Problem you have experienced: I am trying to use the scenario_runner's metrics_manager.py to do something on a recorded log.  I get the following line if I try to print the output of the show_recorded_file_info function in metrics_manager.py
`File /app/scenario_runner/cutout/LaneChangeSimple.log not found on server  `
What you expected to happen: That the metrics_manager would run the way within a docker container the same way it does on the machine. 
Steps to reproduce: 
1. Create a folder named cutout (or whatever you prefer) in the scenario_runner directory.
2. Run `python3 scenario_runner.py --reloadWorld --openscenario srunner/examples/LaneChangeSimple.xosc --record cutout --host host.docker.internal` on the machine.
3. Build a scenario_runner Docker image.
4.  `docker run -it -v path\to\Carla\CARLA_0.9.12\:/app/carla -v path\to\scenario_runner\:/app/scenario_runner --name scenario_test --net host scenario_runner`
5.  `python3 metrics_manager.py --log cutout/LaneChangeSimple.log --metric data_collection.py --host host.docker.internal` will throw an error. 

The issue is similar to the one listed in this [thread](https://github.com/carla-simulator/carla/discussions/4143)

Kin-Zhang, 2022-03-24:
I found that in container, the CARLA recorder didn't work properly.

semmelknoedel, 2022-09-16:
I have the same issue with version 0.9.13. So I mapped the folder from the docker to my system:
`'docker run -d --privileged --name carla --gpus all --net=host -v /tmp/.X11-unix:/tmp/.X11-unix:rw -v ~/carlalog:/home/carla/carlalog carlasim/carla:0.9.13 /bin/bash ./CarlaUE4.sh -RenderOffScreen'`

Then you might still need to make some minor changes to loading the log file in the scenario runner. However, somehow the log-file seems corrupted, since there is no information for the frames. 
`start, end = self.log.get_actor_alive_frames(id)` works fine, but
`self.log.get_actor_transform(id, end)`  (or any other frame) results in the following error:

`Traceback (most recent call last):
  File "metrics_manager.py", line 170, in <module>
    sys.exit(main())
  File "metrics_manager.py", line 167, in main
    MetricsManager(args)
  File "metrics_manager.py", line 61, in __init__
    metric_class(town_map, log, criteria_dict)
  File "path/scenario_runner-0.9.13/srunner/metrics/examples/basic_metric.py", line 29, in __init__
    self._create_metric(town_map, log, criteria)
  File "path/scenario_runner-0.9.13/srunner/metrics/examples/write_data.py", line 28, in _create_metric
    self.build_csv_data()
  File "path/scenario_runner-0.9.13/srunner/metrics/examples/write_data.py", line 58, in build_csv_data
    if self.log.get_actor_transform(id, frame) is not None:
  File "path/scenario_runner-0.9.13/srunner/metrics/tools/metrics_log.py", line 260, in get_actor_transform
    return self._get_actor_state(actor_id, "transform", frame)
  File "path/scenario_runner-0.9.13/srunner/metrics/tools/metrics_log.py", line 194, in _get_actor_state
    frame_state = self._frames[frame - 1]["actors"]
IndexError: list index out of range

Process finished with exit code 0
`