carla-simulator/carla #5288: Issues when "make PythonAPI": fatal: bad revision 'HEAD'

noobC7, 2022-03-27:
- Basic Info:
CARLA version:0.9.12
OS:Ubuntu18.04
Unreal Engine:4.26.2
Python:3.8.12
cmake3.18.3

- Problem:
I followed the instruction in carla document(https://carla.readthedocs.io/en/latest/build_linux/), when I met to "make PythonAPI",  the errors occurred, the details are as follow.

```
(base) cplus7@z2air:~/carla-0.9.12$ make PythonAPI
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c5-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-0b13b0-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
fatal: bad revision 'HEAD'
make: *** [setup] error 128
```
- Could somebody help me? Thanks o lot.

noobC7, 2022-04-03:
- I think it's because of the unstable network connnection. When on stable wifi and make again, the issues disappeared
- Unfortunately, I met another isssue:
'''
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
Util/BuildTools/Linux.mk:87: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1
'''
- Though I didn't find a way to solve this, when I turned to Carla 0.9.11 with UE4.24.3, no issues happened.

leonlee723, 2023-02-07:
run the 'git commit -m "initial commit"' in the terminal. @noobC7 
https://stackoverflow.com/questions/15628720/fatal-bad-default-revision-head

belikeArya, 2023-03-13:
> * Basic Info:
>       CARLA version:0.9.12(git from https://github.com/carla-simulator/carla/tree/0.9.12)
>       OS:Ubuntu18.04
>       Unreal Engine:4.26.2
>       Python:3.8.12
>       cmake3.18.3
> 
>     * Problem:
>       I followed the instruction in carla document(https://carla.readthedocs.io/en/latest/build_linux/), when I met to "make PythonAPI",  the errors occurred, the details are as follow.
> 
> 
> ```
> (base) cplus7@z2air:~/carla-0.9.12$ make PythonAPI
> Setup.sh: llvm-8.0 already installed.
> Setup.sh: boost-1.72.0-c8 already installed.
> Setup.sh: rpclib-v2.2.1_c5-c8 already installed.
> Setup.sh: gtest-1.8.1-c8 already installed.
> Setup.sh: recast-0b13b0-c8 already installed.
> Setup.sh: Libpng already installed.
> Setup.sh: Xerces-c already installed.
> Setup.sh: Sqlite already installed.
> Setup.sh: PROJ already installed.
> Setup.sh: Patchelf already installed.
> fatal: bad revision 'HEAD'
> make: *** [setup] error 128
> ```
> 
>     * Could somebody help me? Thanks o lot.

hello,did you solve the problem yet?
i met the same one too,could you help me if you know how to solve it. thanks very much!

arashfree, 2024-04-03:
the error occurred because of #CARLA_VERSION=$(get_git_repository_version) in line 878 of Setup.sh .
change it to your Carla version and it would be work.
I've changed it to CARLA_VERSION="0.9.15".

CHILLed144, 2025-05-23:
> the error occurred because of #CARLA_VERSION=$(get_git_repository_version) in line 878 of Setup.sh . change it to your Carla version and it would be work. I've changed it to CARLA_VERSION="0.9.15".

it worked for me too, for carla 0.9.13