carla-simulator/carla #5292: generate KITTI format dataset

makaveli10, 2022-03-28:
Hello,
I am trying to create a KITTI format dataset from CARLA. So, to do that I need to build the bounding box from ```vehicle.bounding_box.extent```, ```relative_rotation_y``` relative rotation of vehicle to camera and the loaction of vehicle ```vehicle.get_transform().location``` as done in [here with 0.8.4](https://github.com/Ozzyz/carla-data-export/blob/26c0bec203a2f3d370ff8373ca6371b7eef35300/bounding_box.py).
With reference to [this](https://github.com/Ozzyz/carla-data-export/blob/26c0bec203a2f3d370ff8373ca6371b7eef35300/bounding_box.py), I am able to save the ```midpoint_location``` location as
```
def get_midpoint_from_location(vehicle, extrinsic_mat):
        location = vehicle.get_transform().location
        midpoint_vector = np.array([
            [location.x],  # [[X,
            [location.y],  # Y,
            [location.z],  # Z,
            [1.0]  # 1.0]]
        ])

        sensor_points = np.dot(np.linalg.inv(extrinsic_mat), midpoint_vector)
        return sensor_points
```
where ```extrinsic_matrix = camera.get_transform().get_matrix()```

and for the relative_rotation_y:
```
def get_relative_rotation_y(vehicle, camera):
    """
    """
    rot_vehicle = vehicle.get_transform().rotation.yaw
    rot_cam = camera.get_transform().rotation.yaw
    return (rot_vehicle - rot_cam) * math.pi / 180
```
All the 3 things are saved using [this script]()https://github.com/Ozzyz/carla-data-export/blob/26c0bec203a2f3d370ff8373ca6371b7eef35300/datadescriptor.py to convert to KITTI format( I assumed this is the correct way?)

Now, the issue is to build the box from the available information i.e. ```vehicle.bounding_box.extent```, ```relative_rotation_y```, ```midpoint_from_location``` for which I use the traditional kitti method ```compute_3d_bbox``` where ```P=camera.calibration```
```
def compute_box_3d(obj, P):
    ''' Takes an object and a projection matrix (P) and projects the 3d
        bounding box into the image plane.
        Returns:
            corners_2d: (8,2) array in left image coord.
            corners_3d: (8,3) array in in rect camera coord.
    '''
    # compute rotational matrix around yaw axis
    R = roty(obj.ry)

    # 3d bounding box dimensions
    l = obj.l;
    w = obj.w;
    h = obj.h;

    # 3d bounding box corners
    x_corners = [l/2,l/2,-l/2,-l/2,l/2,l/2,-l/2,-l/2];
    y_corners = [0,0,0,0,-h,-h,-h,-h];
    z_corners = [w/2,-w/2,-w/2,w/2,w/2,-w/2,-w/2,w/2];

    # rotate and translate 3d bounding box
    corners_3d = np.dot(R, np.vstack([x_corners,y_corners,z_corners]))
    #print corners_3d.shape
    corners_3d[0,:] = corners_3d[0,:] + obj.t[0];
    corners_3d[1,:] = corners_3d[1,:] + obj.t[1];
    corners_3d[2,:] = corners_3d[2,:] + obj.t[2];
    #print 'cornsers_3d: ', corners_3d
    # only draw 3d bounding box for objs in front of the camera
    if np.any(corners_3d[2,:]<0.1):
        corners_2d = None
        return corners_2d, np.transpose(corners_3d)

    # project the 3d bounding box into the image plane
    corners_2d = project_to_image(np.transpose(corners_3d), P);
    #print 'corners_2d: ', corners_2d
    return corners_2d, np.transpose(corners_3d)

def project_to_image(pts_3d, P):
    ''' Project 3d points to image plane.
    Usage: pts_2d = projectToImage(pts_3d, P)
      input: pts_3d: nx3 matrix
             P:      3x4 projection matrix
      output: pts_2d: nx2 matrix
      P(3x4) dot pts_3d_extended(4xn) = projected_pts_2d(3xn)
      => normalize projected_pts_2d(2xn)
      <=> pts_3d_extended(nx4) dot P'(4x3) = projected_pts_2d(nx3)
          => normalize projected_pts_2d(nx2)
    '''
    n = pts_3d.shape[0]
    pts_3d_extend = np.hstack((pts_3d, np.ones((n,1))))
    #print(('pts_3d_extend shape: ', pts_3d_extend.shape))
    pts_2d = np.dot(pts_3d_extend, np.transpose(P)) # nx3
    pts_2d[:,0] /= pts_2d[:,2]
    pts_2d[:,1] /= pts_2d[:,2]
    return pts_2d[:,0:2]
```

Thanks in advance.

makaveli10, 2022-03-28:
Tried building the 3d bbox from ```relative_rotation_y```, ```midpoint_location``` and ```vehicle.bounding_box.extent``` obtained using the method mentioned in the above.
```
def get_midpoint_from_location(vehicle, extrinsic_mat):
        location = vehicle.get_transform().location
        midpoint_vector = np.array([
            [location.x],  # [[X,
            [location.y],  # Y,
            [location.z],  # Z,
            [1.0]  # 1.0]]
        ])

        sensor_points = np.dot(np.linalg.inv(extrinsic_mat), midpoint_vector)
        return sensor_points

def get_relative_rotation_y(vehicle, camera):
    """
    """
    rot_vehicle = vehicle.get_transform().rotation.yaw
    rot_cam = camera.get_transform().rotation.yaw
    return (rot_vehicle - rot_cam) * math.pi / 180

extrinsic = ClientSideBoundingBoxes.get_matrix(self.camera.get_transform())

midpoints = [get_midpoint_from_location(vehicle, extrinsic) for vehicle in self.vehicles]
roty = [get_relative_rotation_y(vehicle, self.camera) for vehicle in self.vehicles]
bbox_extents = [vehicle.bounding_box.extent for vehicle in self.vehicles]

def compute_3d_bbox(location, rot, bbox_extent, camera):
        def roty(R):
            c = np.cos(t)
            s = np.sin(t)
            return np.array([[c,  0,  s],
                     [0,  1,  0],
                     [-s, 0,  c]])
        pos = location
        bounding_box_3d = np.array([[pos[0] + bbox_extent.x, pos[1] - bbox_extent.y, pos[2], 1.0],
                     [pos[0] + bbox_extent.x, pos[1] + bbox_extent.y, pos[2], 1.0],
                     [pos[0] - bbox_extent.x, pos[1] + bbox_extent.y, pos[2], 1.0],
                     [pos[0] - bbox_extent.x, pos[1] - bbox_extent.y, pos[2], 1.0],
                     [pos[0] + bbox_extent.x, pos[1] - bbox_extent.y, pos[2] + 2*bbox_extent.z, 1.0],
                     [pos[0] + bbox_extent.x, pos[1] + bbox_extent.y, pos[2] + 2*bbox_extent.z, 1.0],
                     [pos[0] - bbox_extent.x, pos[1] + bbox_extent.y, pos[2] + 2*bbox_extent.z, 1.0],
                     [pos[0] - bbox_extent.x, pos[1] - bbox_extent.y, pos[2] + 2*bbox_extent.z, 1.0]])
        print(bounding_box_3d.shape)
        bounding_box_3d = np.transpose(bounding_box_3d)[:3, :]
        bounding_box_3d = np.dot(R, bounding_box_3d)
        print(bounding_box_3d.shape)
        
        cords_y_minus_z_x = np.concatenate([cords_x_y_z[1, :].reshape((1,8)), -cords_x_y_z[2, :].reshape((1,8)), cords_x_y_z[0, :].reshape((1,8))])
        bbox = np.transpose(np.dot(camera.calibration, cords_y_minus_z_x))
        # print(bbox[:, 0].shape, bbox[:, 2].shape)
        camera_bbox = np.concatenate([bbox[:, 0].reshape((8, 1)) / bbox[:, 2].reshape((8, 1)), bbox[:, 1].reshape((8, 1)) / bbox[:, 2].reshape((8, 1)), bbox[:, 2].reshape((8, 1))], axis=1)
        # print('bbox shape ', coords.shape)
        return coords
```

Using the above method, [client_bounding_boxes.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/client_bounding_boxes.py) and the methods mentioned above you should be able to reproduce the issue.
I think I made some progress as the bbox on the ego vehicle is a little better ?  but its still too bad:
https://file+.vscode-resource.vscode-webview.net/home/makaveli/workspace/carla-dataset/dataset/images_2/image_1.jpg?version%3D1648460927337

closed 2022-05-24 by makaveli10

amsvenkat, 2023-08-04:
Hi. Were you able to solve this ? I have a similar issue, the rotation seems to be off for the objects. WHats the conversion for x,y,z and the rot_y ?

[image]

Petros626, 2025-06-26:
@makaveli10 @amsvenkat any news related to this?