carla-simulator/carla #5300: Issue when running make PythonAPI - not downloading zlib

LeMarcoPaulo, 2022-03-29:
Windows 11
Latest version of CMake and Carla as of the day i'm writing on
CMake added to PATH, using Visual Studio 2019
When I run Make PythonAPI, an error occurs (see picture)
[image]
now I'm not really sure if I put the UE4_ROOT variable in the correct place, maybe that's the problem

if anyone has got a clue, I would appreciate it a lot !

kang-n, 2022-03-29:
I think this is because zlib was updated to 1.2.12, and the linke to 1.2.11 was removed.

ritabt, 2022-04-04:
@kang-n is right. You can fix it by going to carla/Util/InstallersWin/install_zlib.bat and changing line 51 to set "ZLIB_VERSION=1.2.12"
After that delete the Build folder in carla/ and run "make PythonAPI" again

onebitme, 2022-07-21:
> @kang-n is right. You can fix it by going to carla/Util/InstallersWin/install_zlib.bat and changing line 51 to set "ZLIB_VERSION=1.2.12" After that delete the Build folder in carla/ and run "make PythonAPI" again

This solved my issue immediately. Thanks