carla-simulator/carla #5303: Unrealistic weather rendering

mariodoebler, 2022-03-30:
Hi all,

first of all thanks for your great work! Our issue is probably related to  #3494 and #3545. Increasing the cloudiness leads to an "unrealistic" rendering, especially of the sky. Everything looks fine with a cloudiness of 30. Around 35 the sky gets unrealistically dark (till black), even though everything else is still quite bright. We used the provided environment.py for changing the weather parameters. This happens for Carla versions 0.9.11, 0.9.12, and 0.9.13 on Ubuntu 18.04 or 20.04 and Windows, packaged version and built from scratch.

Using Carla version 0.9.9.4, everything looks as expected. I don't see how it could be related to the opengl issue mentioned in the previous issues. Also the rain and fog is rendered correctly, unlike mentioned in some of the previous issues. A contributor even mentioned that opengl isn't supported anymore for the newer Carla versions.

We're more than happy to provide further details, if necessary.

Hoping to find a solution! Thanks!

mariodoebler, 2022-04-08:
We generated a few screenshots to illustrate the problem. The following are sample images by using the provided environment.py, first setting the sun to day, followed by changing the wheather (clear, overcast, rain).

**Carla 0.9.9.4:**
[image]
[image]
[image]

**Carla 0.9.13** (the images are identical for versions 0.9.11 and 0.9.12 and probably, but we haven't tried it, for 0.9.10):
[image]
[image]
[image]

shh1v, 2022-07-18:
Any update on this error?

zzj403, 2023-10-14:
same question. Hope any one gives solution. I try 0.9.14 (-vulkan:bad), 0.9.10.1(-opengl:fine, -vulkan:fine)

shh1v, 2023-10-14:
@zzj403 Take a look at [this ](https://github.com/HARPLab/DReyeVR/issues/45#issuecomment-1189554757) issue. It resolved the problem for me; I don't remember the details, since it was a long time ago.