carla-simulator/carla #5305: wget: unable to resolve host address ‘ftp.cixug.es’

gengtsh, 2022-03-31:
recently, when i run make PythonAPI, i encounter wget issue, 
--2022-03-30 18:30:20--  https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-3.2.3.tar.gz
Resolving ftp.cixug.es (ftp.cixug.es)... failed: Name or service not known.
wget: unable to resolve host address ‘ftp.cixug.es’
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 4

can someone help me out?

RehabHsn, 2022-03-31:
where have you changed the directory? in which file ?

Axel1092, 2022-03-31:
Hi @gengtsh, thank you for noticing this issue. Indeed your solution is the right one. We will add this fix to the repository.

Nikhilds2498, 2022-04-30:
@RehabHsn and @Axel1092  Line number 428 in Setup.sh file  (carla/Util/BuildTools/Setup.sh)

mertbacaksiz, 2022-10-28:
Line number 428 in Setup.sh file XERCESC_VERSION=3.2.3 change to 3.2.4

matejm42, 2022-11-16:
> I am also facing the same error. In which file you have changed https://ftp.cixug.es/apache//xerces/c/3/sources to https://downloads.apache.org/xerces/c/3/sources/ ?

Had the same error, just to sum it up: changing the version just lead to another error.
What worked for me is to change **line 433** in file `Util/BuildTools/Setup.sh` 
from
```
XERCESC_REPO=https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz 
```
to
```
XERCESC_REPO=https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
```

PR #5309 (2022-03-31): Fix xerces download
#### Description

Fixes broken download link for xerces library.

Fixes #5305

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3
  * **Unreal Engine version(s):** 4.26

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/5309)