carla-simulator/carla #5306: Failed to build Boost.Build build engine

martino92, 2022-03-31:
Setup.sh: Extracting boost for Python 3.7.
Building Boost.Build engine with toolset clang... 
Failed to build Boost.Build build engine
Consult 'bootstrap.log' for more details
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1

jdmdj1999, 2022-04-13:
I've got the same problem

Setup.sh: Extracting boost for Python 3.6.
Building Boost.Build engine with toolset clang... 
Failed to build Boost.Build build engine
Consult 'bootstrap.log' for more details
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1

Error in 'bootstrap.log'： 
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)

jiashuowang, 2022-04-13:
me too. +1