carla-simulator/carla #5315: Car crashes to wall with autopilot

Hanss8, 2022-04-02:
CARLA version: 0.9.11
Platform/OS: Windows

Hello, 
i created map in Roadrunner. When i run manual_control.py script and i set autopilot on, my car drives in line for a while, then stops and then drive to side of the road and crashes to wall.
Someone to help me with this issue please? 
Thanks.

makaveli10, 2022-04-04:
Hi, 
Are you using ```synchronous mode``` ?

Hanss8, 2022-04-04:
Hi @makaveli10,
i run the manual_control.py script, as it is after download. So i guess, no synchronous mode is enabled there.
Thank you.

makaveli10, 2022-04-04:
I think you need to turn on synchronous_mode:
```
settings = world.get_settings()
settings.synchronous_mode = True # Enables synchronous mode
settings.fixed_delta_seconds = 0.05
world.apply_settings(settings)
```

shh1v, 2022-07-12:
I am having a similar problem that I posted [here](https://github.com/HARPLab/DReyeVR/issues/41). The vehicle is decelerating after it attains a certain speed, and it also sometimes crashes onto the walking path. Setting synchronous mode does not solve the problem.

closed 2025-08-07 by Blyron