carla-simulator/carla #5316: An error occurred while opening CarlaUE4.exe

hahakuan, 2022-04-02:
# An error occurred while opening CarlaUE4.exe
I get an error when opening CarlaUE4.exe

CARLA version:0.9.13
Platform/OS:windows 11

Here are the logs:
```
Fatal error!

Crash in runnable thread TaskGraphThreadNP 0
Crash in runnable thread TaskGraphThreadNP 3
Crash in runnable thread TaskGraphThreadNP 1
Crash in runnable thread TaskGraphThreadNP 2
```
Hope to solve
Thanks!

Z0184859, 2022-10-11:
i got fatal error while running "Make Launch" command after build.
[image]

can anybody help what we can do ?

Z0184859, 2022-10-11:
it got solved after extracted correctly the assets to the folder "C:\carla\Unreal\CarlaUE4\Content\Carla"

[image]

should look something like this....

thiriveedhisai, 2023-10-26:
can you please elaborate on how did you extract contents?

closed 2024-05-17 by MarcelPiNacy-CVC