carla-simulator/carla #5317: update to latest, make launch fail, directly run ./UE4Editor (pull from CarlaUnreal) also fail

gengtsh, 2022-04-03:
os: ubuntu 20.04,
remote vncserver access 
carla: update to latest version
when run make launch:
two different cases happened: 
1. running for ever, but carla Unreal Engine will not pop up
2. in terminal, it shows LogLinux: could not get DPI information for monitor $0, assuming 1.0f 

when i go to unrealengine folder which i directly pull from "https://github.com/CarlaUnreal/UnrealEngine.git", and directly run ./UE4Editor  
pop out window: 
ExampleDeviceProfileSelector failed to load because module "ExampleDeviceProfileSelector" could not be found, please ensure the plugin is properly installed, otherwise consider disabling the plugin for this project

when i run the UE4Editor which i pull from [github.com:EpicGames/UnrealEngine.git](https://github.com/EpicGames/UnrealEngine.git), there is no any issues. 

can someone help me out, thanks

gengtsh, 2022-04-04:
also encounter error: 
[2022.04.04-23.12.03:946][  0]Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'

[2022.04.04-23.12.03:946][  0]Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'

Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Signal 11 caught.

closed 2022-04-07 by gengtsh

lsangreg, 2022-12-19:
Has someone solved this issue? I have the same identical one.

JohnnyZZhao, 2023-01-22:
Same issue here. Using Ubuntu 20.04.5, Nvidia GeForce RTX2070.

lsangreg, 2023-01-24:
@JohnnyZZhao  I solved the issue by running         ./Update.sh , because I missed this step thinking to  have to download just the assets for a specific version of CARLA:

> also encounter error: [2022.04.04-23.12.03:946][ 0]Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find [...]