carla-simulator/carla #5318: Unreal Engine 5 Roadmap?

RDaneelOlivav, 2022-04-03:
I sthere plans for realeasing an UnrealENgine 5 Version?

Thanks

ll7, 2022-04-21:
I wonder if they changed anything in their Chaos vehicle simulation https://docs.unrealengine.com/5.0/en-US/how-to-convert-physx-vehicles-to-chaos-in-unreal-engine/

junaidbinsarfraz, 2022-10-31:
Still looking for a road map for UE5. It is quite important for us.

Thank you

shott92, 2023-01-20:
Is there any news on the transition to UE5?
I am currently having issues with running UE4 and Carla on ubuntu 20.04 with RX 7900 XT drivers installed.
It seems that I can easily install drivers on new linux distros with linux kernel 6.1.x, but then run into issues with installing the right packages to build carla, which makes sense if 18.04 and 20.04 are the supported versions, maybe it is just support needs adding for ubuntu 22.04 and newer linux kernels.
Does anybody have any thoughts?

brifsttar, 2023-03-07:
Well this just showed up... Lots of work to be done to migrate from PhysX to Chaos, but it looks like it's underway?

https://github.com/carla-simulator/carla/compare/dev...marcel/5.1.1-merge

olegprog, 2023-04-20:
Any news about magrating to UE5 ??? 
I see that this branch from the posts above [marcel/5.1.1-merge](https://github.com/carla-simulator/carla/compare/dev...marcel/5.1.1-merge) is in active development

shott92, 2023-04-20:
@MarcelPiNacy Hi Marcel, how are you doing?
I love seeing the progress towards supporting UE5 on your fork, is there anything we can do to help/support?
would you be able to start a "project" or create an checklist for the issues?
it seems the community are very eager for this and I would be happy to support, as I am sure several others would too!
Thank you kindly

fezsid, 2023-07-21:
Any updates on the roadmap?

shott92, 2023-08-10:
no news as of yet, we've been informed it is progressing, but it appears to be a long way off still, no indications are being made about timings.

There is a relevant fork (https://github.com/carla-simulator/carla/compare/dev...marcel/5.2-merge) however I have tried to build this myself to see where progress is and what next steps might be but i run into several errors, which are difficult to overcome as the documentation and comments aren't updated, such as the docker files suggesting to use ubuntu 18.04.
appropriate Make versions being difficult to find, python and package incompatibilities, against compiler versions.

while this is an open source project, there isn't much in the way of project management to support the co-ordination of community members providing support to the core team

Blyron, 2024-04-12:
Hello! We are happy to announce that we are working on it! We have tons of work to do with a new build system and the transtition to Chaos Physics. 

You can use ue5-dev branch as base for your development 
The new build instructions:
https://docs.google.com/document/d/1aWcoJr-9z_wE6S442UhzC32jUGmN1o42REE7C-4aGYA/edit#heading=h.aodjovbjivi6

closed 2024-04-19 by Blyron