carla-simulator/carla #5326: build Carla 0.9.11 in Windows10, error when executing "make PythonAPI"

albertyxy, 2022-04-05:
CARLA version: 0.9.11
Platform/OS: Windows10
Has anyone similar problem like this when executing "make PythonAPI"?

**The system cannot find the path specified.
    -[install_boost]: Generating build...
'bootstrap.bat' is not recognized as an internal or external command,
operable program or batch file.**

    -[install_boost]: [BOOTSTRAP ERROR] An error ocurred while executing "bootstrap.bat".
    -[install_boost]: Exiting with error...

possible solution: I look up the carla\Util\InstallersWin\install_boost.bat, and find this:
cd "%BOOST_SRC_DIR%"
if not exist "b2.exe" (
    echo %FILE_N% Generating build...
    call bootstrap.bat vc141
)
I think this problem is possibly prompted by not finding the path---cd "%BOOST_SRC_DIR%"
However, I don't know how to solve this. Can Anyone help me with this?

albertyxy, 2022-04-05:
I found this website is broken, so I modified the file install_boost.bat.
set BOOST_REPO=https://dl.bintray.com/boostorg/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE% 
with
set BOOST_REPO=https://boostorg.jfrog.io/artifactory/main/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE%
Is it OK or not? Or is this problem related to this step?

omitheaussie, 2023-10-12:
I found the issue. It is in line 68 in the file install_boost.bat. The zip file that gets downloaded is named boost_1_72_0. The batch script builds the filename string as boost-1_72.0. There are lines (68, 75) that need to be changed to make sure that the BOOST_BASENAME is correct. The path that is built by the batch script is based on an incorrect filename string. So lazily I changed line 110 to hardcode the boost folder:
`cd C:\carla_root\carla-0.9.13\Build\boost_1_72_0`

closed 2024-05-17 by MarcelPiNacy-CVC